arduino/ArduinoCore-samd #235: SDU Referernce missing.

fabltd, 2017-05-09:
The following link is non operational: http://www.arduino.cc/en/Reference/SDU

Is there alternate an alternative SDU reference?

Can the SDU be implmented on any SAMD21 hardware with the Zero bootloader?

sandeepmistry, 2017-05-10:
@akash73 can you please look into the missing reference page. Thanks.

> Can the SDU be implmented on any SAMD21 hardware with the Zero bootloader?

@fabltd only the MKR1000, MKRZero, Zero, MKRFox1200 variants are currently supported. The feature is not tied to the boot loader used on the board, but the linker file used to generate the sketch.

fabltd, 2017-05-15:
Link still broken? Any update ?

agdl, 2017-05-16:
The page is online here https://edit.arduino.cc/en/Reference/SDU

closed 2017-05-16 by agdl

fabltd, 2017-05-16:
Says i need a username and password!

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

agdl, 2017-05-16:
@fabltd sorry I corrected the link

fabltd, 2017-05-16:
Thanks

Why does it need to be a MKRzero can it be any zero board ? Is there any
hardware differences with the MKRzero and a standard zero?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

agdl, 2017-05-16:
You can use a Zero and an SD breakout too

fabltd, 2017-05-16:
Hi

Ok will any SAMD21G18 work if its uploaded as a Zero?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

agdl, 2017-05-16:
Yes because it uses a second stage bootloader. You just have to take care of the SPI used in te variants: i.e for a MKRZero is SPI1 instead for the other boards is SPI.

fabltd, 2017-05-16:
What PINs should the SD card be connected to?

Can the CS be changed?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

agdl, 2017-05-16:
on the MKRZero is the dedicated CS while on the other boards is pin4. You need to recompile the second stage bootloader if you want to change it, so you can't change it out of the box

fabltd, 2017-05-16:
So its PA03? (pin4) - Whats the Git link to the second stage loader?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

agdl, 2017-05-16:
https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SDU/extras/SDUBoot/SDUBoot.ino

anyway you also have this 

https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SDU/examples/Usage/Usage.ino

fabltd, 2017-05-16:
> https://github.com/arduino/ArduinoCore-samd/blob/master/
> libraries/SDU/extras/SDUBoot/SDUBoot.ino
>
> anyway you also have this
>
> https://github.com/arduino/ArduinoCore-samd/blob/master/
> libraries/SDU/examples/Usage/Usage.ino

Thanks

How do i build it? Do i just add the lib and change the parameters?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN