arduino/ArduinoCore-samd #236: UART Bootloader

fabltd, 2017-05-15:
Have implemented the Bootloader and works over USB. Trying to load via Sercom 0. Atmel docs say send # and should get a response. This doesn't seem to work. 

What baud rate should be used is there something missing?

fabltd, 2017-05-16:
Tested UART connection (Serial1) PA10&PA11 in a sketch and TX&RX work.- Sercom0

Have tried two target boards. With boot-loader uploaded. 

Neither will respond to # send over Uart while bootloader is running. (LED pulsing)

facchinm, 2017-05-16:
The bootloader listens on Serial1 if compiled correctly (see https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/main.c#L171 for the needed defines). The protocol implemented is the same as BOSSA over USB (you can use the same `bossac` binary and specifying `-U false`).

facchinm, 2017-05-16:
Ah, and the baudrate should be 115200. Make sure you are using the latest version of the bootloader because otherwise you'll incur in a bug which caused the port speed to be set incorrectly.

fabltd, 2017-05-16:
> bootloaders/zero/main.c#L171 for the needed defines). The protocol

I am using:

https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/board_definitions_genuino_zero.h

This sets the boot-loader as using SERCOM0?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

fabltd, 2017-05-16:
using latest git version.

sam_ba_monitor.h declares:

#define SAM_BA_INTERFACE SAM_BA_BOTH_INTERFACES

So uart should be enabled ?

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

facchinm, 2017-05-16:
Yes. it should be working out of the box on Serial1 (pins 0/1 on Zero board). Did you check with a scope if you can get any response when launching `bossac` on that port (with an USB-to-serial of course)?

fabltd, 2017-05-16:
Hi

I have USB to serial and have simply tried to send # and see if there is a
response. I think this is what bossac does.

I get nothing I have tried everything. I think something is wrong can you
test?

facchinm, 2017-05-17:
I just tested with:
- a Zero with the latest bootloader
- an FTDI cable connected to pins 0 and 1
- power comes from Programming port

Send the board to bootloader mode by clicking the reset button; the onboard LED will start fading. 
Then you can flash with
```
.../packages/arduino/tools/bossac/1.7.0/bossac -i -d --port=ttyUSB0 -U true -i -e -w -v /tmp/arduino_build_128082/Blink.ino.bin -R 
```
It worked at the first try with the default images / IDE tools. Could you perform the same test to make sure your setup is ok?

fabltd, 2017-05-17:
Will test later today.

Is the interactive mode not supported?