arduino/ArduinoCore-samd #237: Running Arduino Zero GCLK Main on OSC8M looks not very accurate

FredMeester, 2017-05-23:
I want to run my Arduino Zero on 8MHz. My configuration of the clocks startup sequence in startup.c:
Step 1: Prescaler to 1 to obtain 8Mhz (SYSCTRL_OSC8M_PRESC_1_Val)
Step 2: GENERIC_CLOCK_GENERATOR_MAIN (GCLK gen 0) using OSC8M

In boards.txt i changed this line: `build.f_cpu=8000000L`
and in variant.h: `VARIANT_MCK (8000000ul)`

For testing, I used a blink LED sketch whit a 3000 ms delay. The duration of this 3s more look like 5s... Am I missing something with the configuration or do I need to calibrate the 8MHz oscillator?

FredMeester, 2017-05-23:
I think a fix was already found by the community... 
https://github.com/arduino/ArduinoCore-samd/commit/4ac0cc1df0f0018791437a4bb498c7ec6bf60edf

sandeepmistry, 2018-01-03:
Closing this a fixed based on the last comment, @FredMeester please let us know if this is still an issue with the latest SAMD core release.

closed 2018-01-03 by sandeepmistry