arduino/ArduinoCore-samd #238: UART Interrupt

Adminius, 2017-06-02:
Is there possible to add a weak function to SERCOM0_Handler and SERCOM5_Handler defined in variants.cpp?
https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/variant.cpp#L212

something like this:

`extern void SerialEvents() __attribute__((weak));
`
```
void SERCOM5_Handler()
{
  Serial.IrqHandler();
  if(SerialEvents) SerialEvents();
}
```

i will call user defined function on serial interrupt (like SerialEvents on AVR), and i can't use another SERCOM (HW-restrictions). So i have to use SERCOM5_HANDLER that is allready defined :/
My own variants.cpp means to musch effort.

Adminius, 2017-06-06:
ok... AVR do it not intrrupt driven.... nevertheless i need it interrupt driven. it will be also ok, if user has an option to override standard handler definition without to edit a variants.cpp file. But i think a weak function is easier way to solve my problem.

Adminius, 2018-04-16:
@facchinm any news? any timeline?
can we make whole SERCOM5_Handler() function weak?
If yes, i can make a PR...

facchinm, 2018-04-27:
Hi @Adminius ,
the sercom "in use" cannot be declared weak (of course, otherwise the empty implementation could be chosen by the compiler) and adding a full SerialEvent ([as in AVR](https://github.com/arduino/ArduinoCore-avr/blob/6e235622edb1b882401c77c171b0e738a093fd5b/cores/arduino/HardwareSerial.cpp#L64)) is not going to happen (we should obsolete it asap).
About your problem, you could quite easily create a `variant` with your needed peripherals so there's no need to change the core or modify the handler at sketch level.
If you need any advice, feel free to comment here and I'll guide you.

closed 2018-04-27 by facchinm