arduino/ArduinoCore-samd #239: Doesn't compile correctly on Arduino M0 and M0+

eljefedelrodeodeljefe, 2017-06-07:
Hi, I can't seem to make the library and it's example run for the the above mentioned boards, running into:
```
InputSerialPlotter:32: error: 'I2S' was not declared in this scope
```

AidenRay, 2017-07-31:
I'm having the same issue with the SAMD21 Xplained Pro board:
`InputSerialPlotter:34: error: 'I2S' was not declared in this scope`
I also tried the AmplitudeSerialPlotter with similar error (but in the c++ file):
`C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoSound\src\AudioInI2S.cpp:36:8: error: 'I2S' was not declared in this scope`

@eljefedelrodeodeljefe any luck?

sandeepmistry, 2017-08-23:
We've added a nicer error message in https://github.com/arduino/ArduinoCore-samd/pull/246 for boards that do not support I2S. Supported boards are: Zero, MKRZero, MKR1000, MKRFox1200.

closed 2017-08-23 by sandeepmistry