arduino/ArduinoCore-samd #247: Error in ArduinoZero flash_with_bootloader.ld - loads at wrong address

frankleonrose, 2017-07-11:
I'm using PlatformIO to build for the Adafruit Feather M0. Using the `flash_without_bootloader.ld` works fine. However, using the `flash_with_bootloader.ld` fails because it does not write the interrupt vector table to 0x0000. Instead, it writes it up at 0x2000.

Looking at https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/linker_scripts/gcc/flash_with_bootloader.ld#L28 it appears that the FLASH origin is set to 0x2000, which is incorrect given that the SECTIONS begins with the bootloader image `KEEP(*(.sketch_boot))`. Shouldn't `.sketch_boot` be written to 0x0000? The line `. = ALIGN(0x2000);` seems to confirm that - `.sketch_boot` should be at 0x0000 followed by the app image at 0x2000. 

```
MEMORY
{
  FLASH (rx) : ORIGIN = 0x00000000+0x2000, LENGTH = 0x00040000-0x2000 /* First 8KB used by bootloader */
  RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 0x00008000
}
```

```
SECTIONS
{
  .text :
  {
  __text_start__ = .;
  KEEP(*(.sketch_boot))
  . = ALIGN(0x2000);
  KEEP(*(.isr_vector))
...
```

facchinm, 2017-07-11:
Hi @frankleonrose ,
the `sketch_boot` section exists only to allow a second stage bootloader to be "plugged" in that section. The 2nd stage has its own vector table at 0x2000 (after the bootloader) and works independently from the sketch. 
If you are not using a library declaring that section, isr_vector is instantiated at 0x2000 as you state, but it's not a problem, since the bootloader already declares it at 0x0000 and jumps to 0x2000 when it launches the user program.
If you don't have a bootloader in place, the `flash_with_bootloader` ld script is obviously wrong. Take extra care when you flash the bootloader + the sketch separately using a programmer, mainline openocd has a bug which erases the first flash page entirely. [This patch ](https://github.com/arduino/OpenOCD/commit/d6cb4bf833a32b87896f1745c01bba594d1cec26) solves the bug and is included in standard samd distribution. I don't know if PlatformIO uses it or another one.
Feel free to reopen if you have any doubt, thanks!

closed 2017-07-11 by facchinm

frankleonrose, 2017-07-11:
Thanks, @facchinm. I'm using BlackMagicProbe, so no need for OpenOCD.

How would `.sketch_boot` ever be anything but zero length given this script?

```
text            0x0000000000002000     0x8894
                0x0000000000002000                __text_start__ = .
*(.sketch_boot)
                0x0000000000002000                . = ALIGN (0x2000)
*(.isr_vector)
```