arduino/ArduinoCore-samd #248: Allow attachInterrupt on NMI pin PA08

ladyada, 2017-07-11:
This will allow IRQs on PA08 - D4 on Zero

[NMI interrupt.txt](https://github.com/arduino/ArduinoCore-samd/files/1139502/NMI.interrupt.txt)

cmaglie, 2017-09-08:
I made a similar patch in the very early stage of samd core, but the NMI is, well, non-maskable :-) it will behave weirdly (for example `noInterrupts` or SPI transactions won't work with it).

Since the SAMD21 has plenty of pin interrupts, that's why I decided to not implement the NMI.

I'll close this one as wontfix, but I'm happy to discuss it further if needed.

closed 2017-09-08 by cmaglie

joverbee, 2017-09-08:
But NMI offers the fastest possible interrupt, which could be important for time critical apps? Or am I wrong? But I agree that it will cause confusion and maybe safer to leave out.

cmaglie, 2017-09-08:
It has a specific NMI_Handler so the timing should a little smaller, but I think that hardly it will make a difference in a real application.

PaulStoffregen, 2017-09-08:
I'd like to share with you a little hindsight about what turned out to be the biggest mistake I ever made with Teensy: exposing Arduino users to NMI.....

PaulStoffregen, 2017-09-08:
The huge problem with NMI is, well, it's always enabled and it has higher priority than all other ARM exception.  "All other ARM exceptions" even includes reset.  If the user happens to connect something that pulls the NMI pin low, the very first instruction executed as the chip comes out of reset can be the first instruction of the NMI handler!

NMI can also interrupt at any point in the startup process, while the chip is still running from its internal RC oscillator and the crystal hasn't started, or the main PLL hasn't locked, or even before the watchdog timer has been turned off.  NMI can cause user code to run before the C runtime startup has copied flash to RAM to initialize all the static & global variables that C code said needed to have initial values.  It can also occur before or during C++ constructors, or even as the startup process initializes the vtables for classes like Stream & Print, which are much of the essence of the Arduino experience.

In the early days of Teensy 3.0 (2012-2013) NMI also caused all sorts of difficult bootloader problems, again because it's not maskable and always enabled.

Fortunately, in the Freescale chip I used, there's something similar to a fuse bit which can completely disable NMI.  The ultimately solution was an update to the bootloader to set that bit as early as possible after chip erase.  Even then, NMI still turned out to cause all sorts of thorny problems.  I'll spare you the rest of those very low level details, only to say it's still gives me nightmares sometimes.

We're all friends, and as your good friend I'd hope this long-winded message can help spare you the pain I and the earliest Teensy users suffered from NMI.

joverbee, 2017-09-08:
I was tempted to try it, but reading this makes me glad I never did :)

cmaglie, 2017-09-09:
@PaulStoffregen thank you very much for sharing your valuable experience (as always!).

winterscar, 2018-03-24:
Sorry to revive an old thread, but I'm working with some hardware that has a sensor attached to pin 4 that I cannot move (It's on a PCB) and I was hoping to watch it with interrupts. All I need to do is set a bool on the rising edge for that pin so there is little danger of complications, but it would seem I cant use attachInterrupt(). 

Could anyone provide me with any info about how I might manually add an ISR to the NMI?

alexwhittemore, 2019-03-29:
So... I may or may not have wired up an i2c to PA08 on a SAMD21E18. PA08=SDA, PA09=SCL. 

Might this be why my request ISRs are always happening before my receive ISRs, when it should be the other way around, meaning the address for the value read isn't set in time?

I'd sure love a hard no or yes, if someone knows enough to give me one :(