arduino/ArduinoCore-samd #250: SAMD21G18A custom board programmed but not executing

SanjeewaUoM, 2017-07-26:
Hi All,

I designed a custom circuit using SAMD21G18A. I programmed it using Atmel ICE. It programs successfully, but nothing executes after that. I debugged the board almost 1 week. VDD, VSS, Reset, RTC is connected correctly. What can be a possible reason?

Regards,
Sanjeewa

facchinm, 2017-07-26:
Hi @SanjeewaUoM , are you uploading a sketch directly without bootloader? In this case, you must change this line (https://github.com/arduino/ArduinoCore-samd/blob/master/boards.txt#L36) (for the board you are targeting, of course) to read `linker_scripts/gcc/flash_without_bootloader.ld` so the sketch is relocated at 0x0000 and is able to boot itself. Hope it helps, please comment here if it works

sandeepmistry, 2017-11-02:
Closing this for now due to lack of feedback. Please re-open with requested info. if you are still interested.

closed 2017-11-02 by sandeepmistry