arduino/ArduinoCore-samd #258: please make 1.0.16 available again.

LPETERS006, 2017-09-02:
So... I switch between two arrays of neopixel to let them Flash. Works so far.... (1.0.16) but since 17....
Sometimes some LEDs show the colors of other LEDs. Sometimes it works correct for hours.
so.., sometimes not really! why is version 16 no longer available? in ArduinoIDE i can get 14,15,17,18... 
16 was the first version where lora and neopixel ran really well together (in my case)
please make 1.0.16 availabe again.
please!!!
Thanks in advance

per1234, 2017-09-02:
Did you maybe post to the wrong repository? There never has been a 1.0.16 release of Arduino SAMD Boards. Even if you meant 1.6.16, well there is no 1.6.17 or 1.6.18...

LPETERS006, 2017-09-02:
Oh, sorry 1.6.16! (0 and 6 look similar) :-)
Good News, Have solved the problem so. 
I've been using eclipse to make a forge on 16. 
Thanks anyway

LPETERS006, 2017-09-02:
Hey ,You were right, probably the wrong forum. However, here is a cut-out from the readme.me of Adafruit 1.0.19:
########################################
## Bugs or Issues

If you find a bug you can submit an issue here on github:

https://github.com/adafruit/ArduinoCore-samd

or if it is an issue with the upstream:

https://github.com/arduino/ArduinoCore-samd/issues

Before posting a new issue, please check if the same problem has been already reported by someone else
to avoid duplicates.

.-)

per1234, 2017-09-02:
OK, I see. The topic is about the "Adafruit SAMD Boards", which apparently did have a 1.0.16 release at one time, rather than Arduino SAMD Boards, which didn't.

In terms of your request to make 1.0.16 available again, that's definitely not an upstream issue (meaning an issue with this repository) so this is not the appropriate place for the issue. I actually think this would be the best place for that request:
https://github.com/adafruit/arduino-board-index/issues
since it's where the change would need to be made.

However, if you wanted to report a bug that's actually causing the Neopixel problem in Adafruit SAMD Boards 1.0.17 and newer you would want to report it to https://github.com/adafruit/ArduinoCore-samd/issues if it's caused by a change Adafruit made to their fork of this repository. Only if the bug was caused by an "upstream issue" meaning a change originally made in this repository should it be reported here.

closed 2017-09-08 by cmaglie