arduino/ArduinoCore-samd #263: problems with M0 bootloader

paul-hoffmann, 2017-09-25:
I used ATMEL STUDIO 7 and an ATMEL ICE SAM to write the M0 bootloader to and Samd. Unfortunally Studio 7 reported the following error

Unable to parse objectfile ...\ArduinoCore-samd-master\bootloaders\mzero\Bootloader_D21_M0_150515.hex: Unsupported format.
 
Same for the bootloader of the M0 PRO

After examining the hex-file I noticed that there were two blank lines at the end. I deleted the last one and everything works fine. 

I do not know anything of the background of this error but it is easy to solve. Can you please change the file.

xseignard, 2019-04-11:
Hello,
I confirm this problem and also the solution provided.
Thanks @paul-hoffmann for reporting it!

nmaas87, 2019-09-18:
Had the same problem as @paul-hoffmann and the analysis was on-point, created a pull request, as the error is also in the official boards-package and you cannot flash a M0 or M0-Pro with this defunct files.
Thanks for reporting!

cmaglie, 2019-09-18:
May you submit a PR with the fix? Thanks!

cmaglie, 2019-09-18:
fixed by #455

closed 2019-09-18 by cmaglie