arduino/ArduinoCore-samd #264: Problems burning M0 bootloader with Atmel-ICE

CarlaMSedze, 2017-10-11:
Hi all,

When trying to burn the bootloader on the Arduino M0 through the Atmel ICE, I get the following error:

"Error while burning bootloader: missing 'erase.params.quiet' configuration parameter"

Arduino IDE 1.8.5
SAMD core 1.6.16

Thank you!

spacepilot, 2017-11-05:
I have the same error Arduino: 1.8.5 (Windows 7), Board: "Arduino M0"

Error while burning bootloader: missing 'erase.params.quiet' configuration parameter

Very few search kits on this issue. Any help would be appreciated.

facchinm, 2017-11-06:
Hi @spacepilot , could you test PR #265 ? If you confirm that it works I'm going to merge it so it will become available in the next SAMD release, thanks!

spacepilot, 2017-11-06:
Hi facchinm, Did as requested with no change.

facchinm, 2017-11-07:
@spacepilot , could you past here the **verbose** output when trying to burn the bootloader? Are you sure you are selecting `Arduino SAMD core - Pull Request #265` from the Boards menu? Thanks!

albertskog, 2017-11-09:
I have the same issue. Tried your PR @facchinm and then I get a new error:

```
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-12:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device

Error while burning bootloader.
```

I might have burnt my chip tho..

facchinm, 2017-11-09:
@albertskog are you on M0 or M0 Pro? You didn't break anything, simply the error message now makes sense (indeed, it doesn't find the programmer). Which is exactly your setup?

albertskog, 2017-11-09:
It's a regular M0, [Crowduino M0](https://www.elecrow.com/wiki/index.php?title=Crowduino_M0-SD) to be precise. It looks like an Arduino M0 to the IDE. The programmer is an Atmel-ICE-Basic ("basic" just seems to indicate what cables comes with it) and I'm running on a Macbook Pro.

The reason I thought I might have killed it was I could not upload programs to it; after pressing reset it went into bootloader mode but then, after the bootloader port went away, no other port appeared, making it impossible to program over usb. I eventually resolved to trying to re-flash the bootloader to see if that would help, that's when I ended up here. Best part was after leaving it disconnected for a while as I was researching this, it suddenly started working again... 😒

closed 2018-05-21 by facchinm