arduino/ArduinoCore-samd #272: SerialUSB.available() returns wrong value/only first byte

scrodev, 2017-11-02:
There is a problem with SerialUSB.available() if using Arduino Zero board.
I send 4 bytes, Terminal output switched from 0 to 4 (but sometimes it only shows 1). Now, if i send another 4 bytes, terminal output dont switch to 8. It still shows only 4.
It seems if available() method executed following received bytes are ignored.

Test sketch
```
void setup()
{
	SerialUSB.begin(115200);
	while(!SerialUSB);
}
void loop()
{
	int availableBytes = SerialUSB.available();
	SerialUSB.println(availableBytes, DEC);
	delay(1000);
}
```

Test procedure:
* burn sketch
* open terminal
* terminal receive a 0 every second
* send 4 bytes
* now terminal receive a 1 or 4 every second
* send another bytes
* terminal should receive 8 but still get a 1 or 4 every second

ScRoDev, 2019-04-23:
Bug is still there

facchinm, 2019-05-28:
Fixed by https://github.com/arduino/ArduinoCore-samd/commit/c8094e7a37bcf70df8c5162bf1bd10e25fb4d5b2

closed 2019-05-28 by facchinm