arduino/ArduinoCore-samd #274: String Descriptor fails to write iManufacturer and iProduct

adammunich, 2017-11-10:
It appears that iManufacturer and iProduct are not filled in for HID descriptor requests. This seems odd, as bool USBDeviceClass::sendStringDescriptor(const uint8_t *string, uint8_t maxlen){} seems to have the strings "Arduino LLC" and "Arduino Zero" passed to it successfully. 

But what's especially strange is the PluggableUSB shortname is successfully set using the same function. 

I suspect the issue may be that the USB string indexes 0x01 and 0x02 are just not being set properly --though I am not fully sure.

adammunich, 2017-11-10:
I fixed the problem;

It seems like in USBCore.cpp, the setup.wlength struct element was not representing the real length of the Product Name and Manufacturing Name descriptor strings. Using sizeof() instead of the struct elements was the problem. 

else if (USB_STRING_DESCRIPTOR_TYPE == t)
	{
		if (setup.wValueL == 0) {
			desc_addr = (const uint8_t*)&STRING_LANGUAGE;
		}
		else if (setup.wValueL == IPRODUCT /* 2 */) {
			return sendStringDescriptor(STRING_PRODUCT, 2*sizeof(STRING_PRODUCT));
		}
		else if (setup.wValueL == IMANUFACTURER /* 1 */) {
			return sendStringDescriptor(STRING_MANUFACTURER, 2*sizeof(STRING_MANUFACTURER));
		}
		else if (setup.wValueL == ISERIAL /* 3 */) {
			#ifdef PLUGGABLE_USB_ENABLED
			char name[ISERIAL_MAX_LEN];
			PluggableUSB().getShortName(name);
		
			return sendStringDescriptor((uint8_t*)name, setup.wLength);
			#endif
		}
		else {
			return false;
		}
		if (*desc_addr > setup.wLength) {
			desc_length = setup.wLength;
		}
	}

I will fork the core and make a pull request

adammunich, 2017-11-10:
Pull request is here:

https://github.com/arduino/ArduinoCore-samd/pull/275

awatterott, 2017-11-11:
The 2nd parameter of `sendStringDescriptor()` is the maximum length and it is `2+sizeof("string")`, because of the 2byte prefix and the string is stored as ACSII (1byte), but send as UTF-16 (2bytes).

adammunich, 2017-11-11:
I accidentally wrote "was the problem", i mean't, "fixed the problem" :-)

The original code uses setup.wlength instead of 2*sizeof("string"), which is incorrect as setup.wlength doesn't always have the right number.

cmaglie, 2017-11-13:
The `setup.wLength` field is the maximum length of the response that should be sent by the micro (even if the descriptor is longer it should be cut at `setup.wLength`, it's the host that controls this field here).
The code IMHO is correct as it is now.

closed 2018-01-02 by sandeepmistry