arduino/ArduinoCore-samd #285: Size of ring buffer limiting size of I2C (and other) messages

lupalby, 2017-12-18:
The Wire library relies on the ring buffer object and the size of the buffer is hard-coded, therefore limiting the size of any outgoing or incoming messages. https://github.com/arduino/ArduinoCore-samd/blob/8f5e2eb58cb91d68c45449b00ed2a7aca794978b/cores/arduino/RingBuffer.h#L28

For instance, there is no limitation to the size of messages in the I2C protocol, other than the available memory in the devices that are communicating. 64 bytes are not much when a lot of data needs to be transmitted, and splitting messages is not always an option.

sandeepmistry, 2017-12-18:
@lupalby any thoughts on https://github.com/arduino/ArduinoCore-samd/pull/269 ?

sandeepmistry, 2018-01-25:
With PR #298 the Wire TX and RX buffers are now 256 bytes in size.

@lupalby please let us know if the new size does not fit your needs and what I2C devices you would like to use.

lupalby, 2018-01-26:
@sandeepmistry Unfortunately 256 bytes is still not enough for my application. I have multiple microcontrollers sending data to each other over I2C and messages can grow up to 1kbyte. With the new solution you guys merged I will still need to go in the Arduino core files and modify the code to bring the size to at least 1000, now in a different location than before.

PaulStoffregen, 2018-01-26:
We're we just talking recently (ok, maybe a year ago) about an addMemory() API for HardwareSerial, Wire, and maybe other libs?

sandeepmistry, 2018-01-26:
@lupalby thanks for the feedback I'll reopen this.

> We're we just talking recently (ok, maybe a year ago) about an addMemory() API for HardwareSerial, Wire, and maybe other libs?

@PaulStoffregen yes, @facchinm proposed PR #233. We need circle back to this at some point.