arduino/ArduinoCore-samd #286: Arduino Zero, Atmel EDBG, and Atmel Studio 7

SurfingDude, 2017-12-21:
I am using the latest Atmel Studio 7 with genuine Arduino.cc Zero boards. When I moved to Atmel Studio 7 it was required that the EDBG firmware update to version 3.25. Some of my boards upgraded fine but some did not, boards bought at different times. The failure was obvious because the upgrade completed immediately, claiming success, but nothing changed.

One of the failing boards was just purchased this month through Arduino.cc and it's clear from the documentation on the web site that EDBG is a feature of the Zero and so it should just work with Atmel Studio.

Clicking on View -> Available Atmel tools then right clicking on EDBG brings up a choice of four options (on the SAMD21XPLAINED board there were 6) and none of the 4 options is "kit mode settings" which would allow selection of "DGI" mode. Discussions in AVRFREAKS forum suggest this setting. Apparently some of the Arduino Zero board hardware comes with improperly initialized EDBG parts.

The ZERO boards that are not debug-able with Atmel Studio 7 work fine with Atmel Studio 6.2.

Is this a production problem at Arduino.cc or is there a way to completely reset the EDBG. It's probably a fuse problem or something in the Arduino supply chain.

SurfingDude, 2017-12-21:
More Clues
==========

On the off chance that it was a driver or registry problem I also tried this on a different Windows 10 computer. The results were the same.

Here is a clue though. In Atmel Studio I used View/Available Atmel Tools. The Zero boards that failed had:
    EDBG (00000000EZE000006092)
and:
    EDBG (00000000AZE000000385)
While the Zero boards boards that worked all had:
    EDBG (FFFFFFFFFFFFFFFF)

Device Manager showed these boards all with the same hardware ID:
    USB\VID_03EB&PID_2157&REV_0101&MI_01
    USB\VID_03EB&PID_2157&MI_01
and I note that VID 0x03EB is listed as Atmel Corporation.

For reference purposes, the working Atmel SAMD21XPLAINEDpro Board had:
   EDBG (ATML2130021800015148)
and:
    USB\VID_03EB&PID_2111&REV_0101&MI_01
    USB\VID_03EB&PID_2111&MI_01

chiefartificer, 2018-01-02:
I have exactly the same situation with my new arduino zero board. It wont upgrade EDGB!

mraardvark, 2018-01-02:
Have you tried using the command line upgrade utility?
atfw.exe -t edbg -a <full path to fw archive>

SurfingDude, 2018-01-02:
Hello Mraadvark,

I guess you are referring to the atfw.exe found in directory:
   Program Files (x86)\Atme\]Studio\7.0\atbackend
but I don't know which zip file it wants and so I am waiting until this gets answered. I aught to be able to use the tool through the USB EDBG port or will I need a JTAG programmer? 

I am temporarily using Atmel Studio 6.2 until Arduino can get this resolved. I am hoping they will be able to fix the problem in their production line so as to not burden any new purchasers.

mraardvark, 2018-01-02:
Sure.
If you want to try: the firmware zip is included in the Studio path \tools\EDBG\edbg_fw.zip for each release of Studio.
In addition, by simply renaming the edbg_fw zip file in that folder, Studio will effectively skip the upgrade check, and continue.
(Mostly new firmware packages are released by Microchip to add DGI features (relevant to XPRO, not Arduino Zero) or new device family support (not relevant to hardware already released), so its unlikely that you need new firmware.)

Fede85, 2018-01-02:
Hi,
I apologize for the delay in the reply.
The Arduino Zero was designed in partership with Atmel (now Microchip) which dedicated to this board a special USB PID and one of the major purposes was to recognize and differentiate form other evaluation boards in Atmel studio.
EDBG ic is used is several Atmel evaluation boards and programmers, you can find the list [here](http://www.atmel.com/webdoc/protocoldocs/ch01s02.html). You should consider the Zero dedicated USB_PID (0x2157) as another USB_PID to add to that list.

During the manufacturing process we upgrade the EDBG firmware and customize the USB descriptor fields. At the moment when a batch is manufactured we flash the latest stable available firmware.
I must check (it's been a while since I tried last time) but, upgrading the firmware with the one provided by Atmel studio using the `atfw.exe` tool will erase all the factory "Arduino" USB configurations and will set the USB_PID to 0x2111. But consider that you couldn't revert the Arduino settings USB descriptor settings.

I think that if the board isn't recognized by Atmel Studio the issue is on Microchip/Atmel side, and you try to contact their support too. I'll do as well.

I need a bit of time to try but I can upgrade Atmel Studio and the EDBG firmware and let you know what you can expect.

The procedure and advice suggested by @mraardvark are correct!

chiefartificer, 2018-01-06:
Several days ago I did rename the EDBG folder to "_EDBG" and debugging now works with Atmel Studio 7 and the firmware version installed on the Arduino Zero.  For the time being I don't want to risk erasing the Arduino USB configuration.

johvdrie, 2018-03-14:
Same problem here with my Arduino genuino zero board. Although the workaround does a good job, is there a final solution ?