arduino/ArduinoCore-samd #290: Update release version of arm-none-eabi-gcc

xloem, 2018-01-08:
Arduino currently provides arm-none-eabi-gcc tool version 4.8.3 .

This version is very old and is no longer distributed at arm.com .

If the compiler were upgraded to at least version 6, then modern c++ tools like boost::hana could be used.

This issue was moved from https://github.com/arduino/Arduino/issues/7088 .  I understand upgrading the toolchain could be a big effort because it needs to be validated again.  It would be great to have a note regarding what this validation process is, in case collaborators could help.

stevenj, 2018-05-20:
Seconded. 4.8.3 is ridiculously old and some software (eg, FreeRTOS 10) won't build with it.

matthijskooijman, 2018-05-29:
I think this one can be closed in favor of #313 (which requests the same, but has a bit more discussion already).

facchinm, 2018-06-04:
Closing ad duplicate and tracking on #313 , staging packages are due to be available later today.

closed 2018-06-04 by facchinm

facchinm, 2018-06-04:
By adding http://downloads.arduino.cc/packages/package_samd_newtoolchain_index.json you can test the latest toolchain; please test and report any error you may encounter, thanks!

facchinm, 2018-06-05:
@stevenj yes, it's a known issue (see https://github.com/arduino/ArduinoCore-samd/issues/313#issuecomment-394286400). Please continue the discussion there so we don't split up the topic :stuck_out_tongue: