arduino/ArduinoCore-samd #293: SerialUSB cause memory leak after LowPower.sleep(), using ArduinoLowPower lib

Terryp44, 2018-01-15:
Every wakeup causes some memory loss, eventually causing CPU reset. However if USB is not plugged in, no memory is lost.
using USBDevice.detach(); and USBDevice.attach();, open/close port does not help.
Using samd version 1.6.17
Test RAM memory usage via:
extern "C" char *sbrk(int i);
int FreeRam (void)
{
  char stack_dummy = 0;
  return &stack_dummy - sbrk(0);
}

Terryp44, 2018-01-17:
After reconnecting the SerialUSB port to the PC. samd CPU ram memory decreases by 592 bytes every sleep cycle.

facchinm, 2018-01-17:
Hi @Terryp44 ,
I pushed #295 to solve the issue. Can you test the produced json and confirm that the standby "dance" doesn't leak anymore? Thanks!

closed 2018-01-25 by sandeepmistry