arduino/ArduinoCore-samd #299: Compile error when using variable name "index"

basevers, 2018-01-25:
When instantiating a variable named "index" causes a compile error

Arduino: 1.8.5 (Windows 10), Board: "Arduino MKRZERO"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bas\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bas\Documents\Arduino\libraries -fqbn=arduino:samd:mkrzero -ide-version=10805 -build-path C:\Users\Bas\AppData\Local\Temp\arduino_build_731251 -warnings=all -build-cache C:\Users\Bas\AppData\Local\Temp\arduino_cache_834896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -verbose C:\Users\Bas\Documents\Arduino\myWebUSB\myWebUSB.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bas\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bas\Documents\Arduino\libraries -fqbn=arduino:samd:mkrzero -ide-version=10805 -build-path C:\Users\Bas\AppData\Local\Temp\arduino_build_731251 -warnings=all -build-cache C:\Users\Bas\AppData\Local\Temp\arduino_cache_834896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -verbose C:\Users\Bas\Documents\Arduino\myWebUSB\myWebUSB.ino
Using board 'mkrzero' from platform in folder: C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17
Using core 'arduino' from platform in folder: C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17
Detecting libraries used...
"C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_MKRZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804f -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKRZero"' "-DUSB_VERSION=0x210" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\cores\arduino" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\variants\mkrzero" "C:\Users\Bas\AppData\Local\Temp\arduino_build_731251\sketch\myWebUSB.ino.cpp" -o "nul"
"C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_MKRZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804f -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKRZero"' "-DUSB_VERSION=0x210" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\cores\arduino" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\variants\mkrzero" "-IC:\Program Files (x86)\Arduino\libraries\WebUSB\src" "C:\Users\Bas\AppData\Local\Temp\arduino_build_731251\sketch\myWebUSB.ino.cpp" -o "nul"
"C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_MKRZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804f -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKRZero"' "-DUSB_VERSION=0x210" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\cores\arduino" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\variants\mkrzero" "-IC:\Program Files (x86)\Arduino\libraries\WebUSB\src" "C:\Program Files (x86)\Arduino\libraries\WebUSB\src\WebUSB.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_MKRZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804f -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKRZero"' "-DUSB_VERSION=0x210" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\cores\arduino" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\variants\mkrzero" "-IC:\Program Files (x86)\Arduino\libraries\WebUSB\src" "C:\Users\Bas\AppData\Local\Temp\arduino_build_731251\sketch\myWebUSB.ino.cpp" -o "C:\Users\Bas\AppData\Local\Temp\arduino_build_731251\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Bas\AppData\Local\Temp\arduino_build_731251\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\Bas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -Wall -Wextra -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_MKRZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804f -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKRZero"' "-DUSB_VERSION=0x210" "-IC:\Users\Bas\AppData\Local\Arduino15\packages\ardu [... cut ...]
                 from C:\Users\Bas\AppData\Local\Temp\arduino_build_731251\sketch\myWebUSB.ino.cpp:1:

c:\users\bas\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\string.h:55:8: error: previous declaration of 'char* index(const char*, int)'

 char  *_EXFUN(index,(const char *, int));

        ^

Using library WebUSB at version 1.0 in folder: C:\Program Files (x86)\Arduino\libraries\WebUSB 
exit status 1
'int index' redeclared as different kind of symbol

facchinm, 2018-01-26:
Indeed, `index()` is a function exported by libc (like `malloc` or `free`) if `__STRICT_ANSI__` is not defined https://github.com/eblot/newlib/blob/master/newlib/libc/include/string.h#L55 so it can't be used to name a variable.

closed 2018-01-26 by facchinm

basevers, 2018-01-26:
Awesome. Thanks for the fast response.