arduino/ArduinoCore-samd #302: USBDeviceClass::send fails with payloads larger than 127 bytes.

markmal, 2018-02-03:
Hi!
I am working on USB MSC device. I use PluggableUSBModule, and everything is quite good, except...
My program correctly registers Bulk SCSI endpoints and correctly responses to OS driver SCSI requests, such as Inquiry, Test Unit Ready, Read Capacity etc, all that required for MSC SCSI device.
However when data sent in response to Read10 request, it sends only 127 bytes, and then it fails right after that.

pseudo-example:
`uint8_t data[4096];
uint32_t len=4096;
uint32_t rl = USBDevice.send(InDataEndpoint, data, len);
/*** no output after this point, device hangs ***/
`
please advise.

facchinm, 2018-02-05:
Hi @markmal ,
this may be related with ZLP handling. Does it work if you send the data in 127 bytes chunks?

markmal, 2018-02-05:
Hi facchinm,
chunks do not work. MSC host expects all in one chunk on protocol level. 
However I've looked into code, low level chunking by <=64 bytes is done in USBDeviceClass::send(uint32_t ep, const void *data, uint32_t len); in USBCore.cpp

facchinm, 2018-02-05:
Would you mind sharing the minimum necessary sources of your project so I can try reproducing it? Testing with CDC doesn't show any strange behaviour so there must be something different going on.

markmal, 2018-02-05:
just created git project
https://github.com/markmal/MSD

facchinm, 2018-02-06:
Hi Mark,
I tested your library and hit a different wall; instantiating an `MSC_` object alone is sufficient to hang the board completely. I'd really like to help you but I need some working code to get started (at least the enumeration stuff). Thanks

markmal, 2018-02-06:
Hi facchinm, thanks for trying to help! 

Instantiating of MSC_ works for me. As well as everything up to SCSI read requests. All [revious communication with USB and SCSI driver works well in my case. It fails only when host requests data and data larger than 127  sent. Even read requests work if no data sent.

CDC alone works well. Though it also fails after MSC sends data. Whole USB in board renders unconfigured due to reset interrupt caught. I do not understand why and what sends this interrupt.

My board is Adafruit Feather M0, may be some pinouts are different? 
I develop in Linux. I read that Windows sends some undocumented (from MSC SCSI perspective) multimedia requests (i.e. READ_FORMAT_CAPACITY), that I do not process yet. May be this happens in your case?
What board, OS, dev env do you use?

Also I've found that layout of members in class may cause hangs. ARM needs them aligned strictly according to their size. May be there is some misalignment somewhere. Do you know how to make compiler to warn on such misalignment?

markmal, 2018-02-06:
I've compiled and run the code in Windows 7. It recognizes my MSC drive, I can see it in Device Manager. Of course it cannot be opened coz all read requests return ABCDE... garbage. Windows just tries and tries to open it, that makes other apps kind of waiting. 

I use Eclipse with PlatformIO in Windows too. No issues with compilation and upload. and the board runs (blinks in loop).
I've refreshed files in github. 
I've made the project a library, in PlatformIO project it is located in <projectdir>/lib/MSD.
I had to move main.cpp outside, into <projectdir>/src 
git is just for MSD, so no main.cpp is there. I will place it somewhere like /test or so.

facchinm, 2018-02-07:
Hi @markmal ,
I had a closer look and indeed all the problems are RAM related. All the debug prints are smashing the stack after a few dummy iterations; I removed them to enable further debugging and found that the problem has already been solved by this commit https://github.com/arduino/ArduinoCore-samd/commit/2bb54068a2ffad535d9d908f0bdc4ec0f1d46708 .
At least, using `master` branch I'm not able to reproduce the hang. To test against `master`, simply follow [these](https://github.com/arduino/ArduinoCore-samd#hourly-builds) instructions

markmal, 2018-02-07:
THANK YOU MARTINO!

markmal, 2018-02-07:
Though... it still does not work in my case. Before applying the patch  2bb5406, I've checked USB/USBCore.cpp, it had exactly what the patch should change. Because previously I've updated everything from github release 1.6.17.
I also cleaned up all Serial.prints and even debug+="..." outputs. 
Even do not do Serial.begin().
Does not help. :(

Could you please share your variant of code that works?

facchinm, 2018-02-08:
Hi Mark,
I'm not really into platformIO or any other alternative platform, but are you sure you are compiling against git master branch? Double check the verbose compilation paths. I'm attaching the modifications I did to your library, nothing fancy, I only commented the debug stuff, leaving the fake SD code intact. Since I could reliably reproduce the problem before applying the patch and I saw no hang after I'm pretty sure this was solve by it.
I'm sorry I can't give you further assistance but the project is cool enough to gain a lot of traction on the [forum](forum.arduino.cc), so if you need some help you can try writing there.

[0001.disable-debug.zip](https://github.com/arduino/ArduinoCore-samd/files/1706265/0001.disable-debug.zip)

markmal, 2018-02-08:
Thank you Martino again!
I'll double-check the paths. 
Also I am separately compiling the lib in Arduino IDE too.

markmal, 2018-02-09:
In my case sending big data started working after I've replaced in USBCore.cpp 
uint32_t USBDeviceClass::send(uint32_t ep, const void *data, uint32_t len)
from
`if ( len >= EPX_SIZE ) {
			length = EPX_SIZE - 1;
		} else {
			length = len;
		}`
to
`if ( len > EPX_SIZE ) {
			length = EPX_SIZE;
		} else {
			length = len;
		}`

facchinm, 2018-02-12:
Hi @markmal , that code was last modified in https://github.com/arduino/ArduinoCore-samd/commit/40d9554ed333c363fee2f384295b8bcf800a6a97 to avoid sending a ZLP in case the data to be sent is a multiple of the EP size. Probably there is a bug lurking somewhere, so thank you for taking a look at it.
@cmaglie should we backport https://github.com/arduino/ArduinoCore-avr/commit/3b0765979d6ed1b689e65381e264c34b071a01b4 to samd?

cmaglie, 2018-02-13:
Probably yes, I'm wondering if the samd's autoZLP feature may help here.

markmal, 2018-02-13:
Thank you Martino and @cmaglie!
So far I've made big progress, device works in Linux quite well as read-only. 
Will do WRITE10 and test in Windows7 soon.

BTW, Martino the forum link https://github.com/arduino/ArduinoCore-samd/issues/forum.arduino.cc returns "Not Found"

facchinm, 2018-02-13:
:+1: I've updated the link, it was simply https://forum.arduino.cc

markmal, 2018-02-14:
I've made the write operation. But it is not very stable.
USB speed read and write is quite slow (~200KB/s).
and it hangs after receiving couple of MB... :(