arduino/ArduinoCore-samd #304: USBH_MIDI driver ported

gdsports, 2018-02-15:
Any interest in the USBH_MIDI driver from the USB Host Shield 2.0 project? I can submit a PR. Would it be better to keep the USBH_MIDI code in a separate repo or include with the core files?

https://github.com/gdsports/usbh_midi_samd

I have not done much testing except with a MIDI mixer board.

bonnom, 2018-04-17:
I came here to request this. It would be great if it gets integrated in the core!!

gdsports, 2018-09-13:
The above repo is no longer maintained. I ported most of the drivers including the MIDI host driver from the USB Host Shield 2.0 project to work on SAMD. Early versions required changes to the SAMD core files but the current version does not.

https://github.com/gdsports/USB_Host_Library_SAMD/