arduino/ArduinoCore-samd #305: Unsigned driver warning on Windows 7

PaulStoffregen, 2018-02-16:
When installing version SAMD version 1.6.17 into Arduino 1.8.5 on a fresh Windows 7 system, two of the driver installs give the red "Windows Security" warning.

[image]

facchinm, 2018-02-16:
Mmmmh, that's bad... Even if the certificate **may** be expired, we surely signed the drivers before the expiration date... @cmaglie any hint?

cmaglie, 2018-02-16:
oh no again... this is a nightmare... will check this out in the next days...

PaulStoffregen, 2018-02-16:
Maybe it's the SHA1 issue?

awatterott, 2018-02-18:
I use also Win7 and I get only an error for the Adafruit CPX driver. The Arduino and Genuino SAMD drivers install correctly.

dlabun, 2018-02-21:
On my computer it's always the Adafruit drivers that throw the security errors. I've told Adafruit numerous times that their drivers aren't signed properly.

cmaglie, 2018-07-25:
The lastest 1.6.19 should have this problem fixed.

Please comment further if this is not the case...

closed 2018-07-25 by cmaglie