arduino/ArduinoCore-samd #309: Changes to attachInterrupt() prevent pin set-up with NULL callback pointer

MartinL1, 2018-02-18:
Changes to the attachInterrupt() function prevents the pin from being set-up when a NULL pointer callback is detected.  

The older attachInterrupt() code allowed for both, enabling the attachInterrupt() to be used for setting up events on the SAMD21's event system, with the callback argument being passed as NULL or 0. This is now not possible.

The event system can be used to allow pin interrupts to trigger TCC timer capture, please see Arduino forum thead: [https://forum.arduino.cc/index.php?topic=396804.0](url). The change to attachInterrupt() break the code examples on this thread.

The workaround is to use a dummy callback in place on the NULL, this solution however is rather inelegant.

facchinm, 2018-02-19:
This is the related code change (https://github.com/arduino/ArduinoCore-samd/commit/9a49ce2fc4d4944dc84c33778339306b092816c8#diff-2d6f462f44c0386e70bfb9e3ba804d5bR84). I must admit that I agree with @MartinL1 , the SAMD21 case is quite particular but not unique, so we should revert part of the patch (leaving the null check but configuring the interrupt anyway) @cmaglie

MartinL1, 2018-02-19:
Thanks @facchinm, 

Just also like to say thank you to all the developers working on the SAMD21 Arduino core, much appreciated.