arduino/ArduinoCore-samd #310: I2S sampling freq not correct

andrewjfox, 2018-03-02:
Hi,

I've been trying to get audio from a modem using the I2S/PCM interface and having some troubles getting accurate clock frequencies.

I'm trying to read single 16bit words from modem @ 8kHz, by configuring the I2S as 8bit slots, since I see in the library it's expecting stereo so has min 2 slots. Following is code to init device, note that I am adjusting clock polarity and FS width to suit modem. 
```
 I2S.onReceive(ISR_onI2SReceive); 
 
  if(!I2S.begin(I2S_PHILIPS_MODE, 8000, 8)) return false;  
  
  /*  SAMD21 I2S peripheral samples data on +ve clock edge, however modem data is valid on -ve edge
   *  if SAMD is master then can invert output clock, this only changes output, so internal operation is unchanged
   *  modem requires FS to be 1 bit width (in slave mode)
  */
  i2s_disable();
  REG_I2S_CLKCTRL0 &= ~I2S_CLKCTRL_FSWIDTH_Msk;  //clear FS width
  REG_I2S_CLKCTRL0 |= I2S_CLKCTRL_SCKOUTINV | (I2S_CLKCTRL_FSWIDTH_BIT_Val << I2S_CLKCTRL_FSWIDTH_Pos);  //invert clock output, FS 1 bit wide (TDM mode)
  i2s_enable();

```
For this setup, with 8kHz sample rate I would expect SCLK to be 8000 * 2 * 8 = 128kHz, however when I monitor the I2S lines using logic analyser, the clock is around 400kHz, and FS is around 25kHz. See attached screen capture,
[image]
 
If I have it configured for 8kHz and 16bit data, then clock and frame are at 256kHz and 8kHz resepctively, as expected.

I had a look at the driver code but the clock setting is a bit complex at first glance so wasn't really able to locate where the issue is.

Regards
Andrew
Hope this is

andrewjfox, 2018-03-08:
Hi @sandeepmistry , just bumping this to see if I can get a response, and I think you've been involved in the I2S stuff before.

andrewjfox, 2018-03-14:
Please! Is anyone able to validate that this is a bug in the I2S library?

sandeepmistry, 2018-03-20:
The library is fine, I tried the simple tone example with 8 kHz and 16 kHz:

[image]

[image]

Since you made modifications, I suggest you discuss on the forum or create pull request to get your suggestions included in the core.

andrewjfox, 2018-03-20:
Did you read the post! I didn't make modifications to the library. I added an extra call, in my own code, to the registers to change the frame sync bit width. And I didn't say changing sampling freqs didn't work, I clearly stated that changing word width from 16 bits to 8 bits caused problem.

andrewjfox, 2018-03-22:
@sandeepmistry this is not closed.

I try this code,
``
#include <I2s.h>

#define I2S_SAMPLERATE_HZ     8000
#define I2S_WORD_BITS         8

void setup() {
  I2S.begin(I2S_PHILIPS_MODE, I2S_SAMPLERATE_HZ, I2S_WORD_BITS);  
}

void loop() {
 if(I2S.available()) I2S.read();    
}
''

and I get following result

[image]

I haven't modifed library, I only changed bit width to 8 bits in I2S.begin
If you change bit width back to 16 it works as expected.

sandeepmistry, 2018-03-22:
@andrewjfox thanks for the sketch to reproduce, I can reproduce the problem with it.

Calculations
```
48000000 / (8 * 2 * 8000) = 375 
```
is too big to fit in the 8-bit divider value, so gets truncated to 119. If you use 8-bit @ 16kHz or 16-bit @ 8 kHz the issue does not occur. Just 8-bit @ 8 kHz

sandeepmistry, 2018-03-22:
Please provide your feedback on pull request https://github.com/arduino/ArduinoCore-samd/pull/317 and let us know if it fixes the sketch from https://github.com/arduino/ArduinoCore-samd/issues/310#issuecomment-375144902. Thanks.

andrewjfox, 2018-03-22:
Yep, that's fixed it. Thanks for the quick reply.

sandeepmistry, 2018-07-31:
Closed via #317.

closed 2018-07-31 by sandeepmistry