arduino/ArduinoCore-samd #321: Low-power Sleep doesn’t work after programming - SAMD21

arrowmax08, 2018-05-14:
I have an issue with my SAMD21 board. The board consumes around 40uA in SLEEPDEEP mode. Only, after reprogramming the board, it consumes around 9mA. A lot more... The reset cause after reprogramming the board is 64, System Reset Request.

The low power sleep code:

```
setup(){
   // Sets SLEEPDEEP
   SCB->SCR |= SCB_SCR_SLEEPDEEP_Msk;
} 

loop(){
   // Put SAMD in sleep
   __WFI()
}
```

The board goes back in sleep when the battery is unplugged/plugged back in. I can't figure out why the board doesn't sleep after programming. 

Any help?

facchinm, 2018-05-16:
Hi @arrowmax08 ,
which D21 board are you referring to? The cause may be the USB initialization that is performed during upload, which doesn't happen if the board is already programmed.
May it be the case?

arrowmax08, 2018-05-16:
Hi Martino, It is a custom board powered on battery. We program the board using Serial Wire Debug (Atmel-ICE programmer)

rocketscream, 2018-07-08:
@arrowmax08 you will need to do proper USB.detach() before sleeping in order to achieve low uA. Even at 40 uA, that is high.

ldab, 2019-04-05:
Not sure if relevant because very old topic but may help someone looking for it.

`wiring.c` set all Pins as Inputs, changing it to `INPUT_PULLUP` reduced my sleep current significantly.