arduino/ArduinoCore-samd #334: Created a new core for the SAMD20 Processor

warrenwoolseyiii, 2018-07-21:
Hi All,
I am unsure where to put this or who to get in contact with so I figured I'd just post as an issue here. I have created a new Arduino code for the SAMD20 processor family (currently supporting the SAMD20E018 but have future plans to expand to the G variant as well). The D20 is pretty comparable to the D21 but there are some pretty big differences. I am unsure if I should add my library here as a "variant" with a bunch of compilation switches in the source files, or if I should create a new repository for all D20 variants. Any advice or direction towards standard protocol for this would be appreciated :)

cmaglie, 2018-07-26:
What are the main differences?
Have you started/forked from the existing SAMD21 core and added the needed functions to support D20?

Would be nice if we can see a comparison between the two cores and see if it's worthwhile to merge them trough a pull request.

On the other hand, there is a risk that adding support for a CPU that we don't sell would likely lead to that CPU being poorly maintained.

warrenwoolseyiii, 2018-07-27:
Originally I forked this repo to start with. The main differences between the D21 and the D20 families is that the D21 has the USB peripheral, and the D20 does not. They also have different register structures, and a different synchronization mechanism between peripherals and the synchronous clock bus.

The changes I have made to the core have aimed it at a lower power usage case than the D21 core, I have found that in the deepest sleep mode I am achieving just 3.9 uA of current consumption by the processor. I had to change things like the systick mechanism so that the processor wasn't waking up to service it every tick, instead I am using the RTC module and a pretty cool timer method that pauses and un-pauses during sleeps for higher resolution timing (micros). I have also added some other core functions that would be useful, like the ability to change the CPU clock speed and source at run time. I have also added a TimerCounter module that allows easy PWM and interrupt based timer code to be used in a similar fashion to the external interrupt controller already present in the core. I also added an arduino style EEPROM module to the core called EEEPROM. The D20 has a bank of flash memory reserved for EEPROM emulation, so I built an emulator that uses a load balancing algorithm to ensure we don't burn out a single flash bank to quickly.

Thinking about this out loud, I think it may be pretty drastic changes to the core which would be too much for a pull request because I have had to touch just about every file. I would be happy to open a pull request though so everyone can see the differences. I am actively working on the core at this time, I still have to make an Arduino compatible bootloader and then figure out what support files I need to create to add this as a variant to the IDE. My fork is on my personal repository located here:

https://github.com/warrenwoolseyiii/Arduino-SAMD20

At the moment, many of the pushes are just internal for me and I haven't documented anything in great detail yet however I do plan to do significant work on a README in short time. As for maintaining the code etc, I was hoping that by posting here I could eventually have this repo live on the Arduino GitHub so that anybody could contribute to this core. Speaking only for my self here, I plan to actively work on this project for at least the next year as it is tied to another project I have to do regardless.

robertfoss, 2018-12-09:
I very much would like to see something like this merged.

warrenwoolseyiii, 2018-12-10:
Hi @robertfoss, I can definitely work on getting a merge. But I made some pretty hefty modifications to many of the core files, and added some additional ones (like support for the RTC) different sleep / low power modes, modifications to millis(), micros() to work with the RTC and work around the deepest sleep modes of the processor. With all of these changes it will probably be a pretty ugly merge and I have to test it on a samd21 to make sure I haven't broken any original core stuff. I will open an issue on my repo and start working towards this. Please feel free to fork my repo or clone it in the mean time if you would like to use this on a samd20!

warrenwoolseyiii, 2019-02-05:
Hey All,
Just wanted to update everyone. I have forked this repo and I am currently in the process of putting together a PR. Its going to take some time because I have got to test all of my changes on the samd21 and the samd20 boards. But looking forward to hopefully expanding support to the samd20 processors!

robertfoss, 2019-05-09:
@warrenwoolseyiii How is the fork? Is there a branch of it that I can have a look at?

sandeepmistry, 2019-09-16:
I'm closing this for now, there is no plans to support SAMD20 in this core or create a new official core.

closed 2019-09-16 by sandeepmistry