arduino/ArduinoCore-samd #336: Change in Windows Drivers breaking Circuit Playground Express Bootloader

ladyada, 2018-07-26:
Hey @cmaglie 

Several people have recently had trouble using CPX on Windows 10: CPLAYBOOT disk drive doesn't show up. This seems to happen after they install Arduino.
Example: https://forums.adafruit.com/viewtopic.php?f=58&t=138649

We know from experience that you shouldn't install the .inf file on windows 10, and let windows 10 use the default CDC driver. 

What's odd is that this wasn't an issue till recently, did the install procedure change?

cmaglie, 2018-07-27:
On the IDE side the installation process is not changed, BTW we released the 1.6.19 with some changes regarding the drivers:

https://github.com/arduino/ArduinoCore-samd/commits/master

in particular this two:

1) We added some new boards
https://github.com/arduino/ArduinoCore-samd/commit/2b39158477739746033c81f62cfdecea365e7796

2) we re-signed your drivers because they [were showing red-alert boxes](https://github.com/arduino/ArduinoCore-samd/issues/305):
https://github.com/arduino/ArduinoCore-samd/commit/de0fa4f32dd3c787044fe044b4f0872d50db6add

Now I'm wondering if the commit n.2 affects your users is some way?
The only thing that comes to mind is that the wrong signature was forcing the driver to not be installed in the system?

ladyada, 2018-07-27:
not sure...but its definitely new since the last update. happened to me as well :) can you set it so it doesn't install on windows 10? our driver installer exe has a check and doesnt install on win10:
https://github.com/adafruit/Adafruit_Windows_Drivers/blob/master/adafruit_drivers.nsi#L130

cmaglie, 2018-07-27:
We use the post_install.bat file: https://github.com/arduino/ArduinoCore-samd/blob/master/post_install.bat

There is a way to check for Win10 in .bat file?

ladyada, 2018-07-27:
according to https://stackoverflow.com/questions/13212033/get-windows-version-in-a-batch-file#13212088

@echo off
```
setlocal
for /f "tokens=4-5 delims=. " %%i in ('ver') do set VERSION=%%i.%%j
if "%version%" == "10.0" echo Windows 10
if "%version%" == "6.3" echo Windows 8.1
if "%version%" == "6.2" echo Windows 8.
if "%version%" == "6.1" echo Windows 7.
if "%version%" == "6.0" echo Windows Vista.
rem etc etc
endlocal
```

dhalbert, 2018-07-27:
@cmaglie I don't think either of the recent commits you made would cause an issue. It's possible that the latest major Windows update did something that's affecting driver installs, though it's not clear to me what it that might be. It could also be just a coincidence that we are seeing users reporting this problem more recently.

In our latest drivers package, https://github.com/adafruit/Adafruit_Windows_Drivers/blob/master/Drivers/Adafruit_usbser/Adafruit_usbser.inf#L205, we specify the `MI` for everything (see below). This might make the driver work successfully on Windows 10 and not hide `CPLAYBOOT`. I tried testing an modified version of the `.inf` from the BSP but I need to sign it for it take priority over the default one. I have signing tools so I'll try that and report back.
```
"%CIRCUIT_PLAYGROUND_EXPRESS% UF2 Bootloader (0018:00)"=DriverInstall, USB\VID_239A&PID_0018&MI_00
"%CIRCUIT_PLAYGROUND_EXPRESS% UF2 WebUSB dummy (0018:04)"=NullInstall, USB\VID_239A&PID_0018&MI_04
"%CIRCUIT_PLAYGROUND_EXPRESS% Arduino (8018:00)"=DriverInstall, USB\VID_239A&PID_8018&MI_00
"%CIRCUIT_PLAYGROUND_EXPRESS% CircuitPython (8019:00)"=DriverInstall, USB\VID_239A&PID_8019&MI_00
```

dhalbert, 2018-07-30:
Filed a PR to fix this. @cmaglie could you make a new minor release after merging? Thanks!

PR #338 (2018-07-31): CPX fixes: add &MI_00 to all devices so CPLAYBOOT is not hidden on Windows 10
File naming has also been normalized to match the other drivers in this directory: all lower case, and the `.inf` and `.cat` files now have the same name.

Fixes #336.

@ladyada and I have tested this on Windows 7 and Windows 10.

Could you make a new minor release after accepting this PR? Thanks!

(EDIT: also version number of driver has been bumped and made more rational)
closed 2018-07-31 by cmaglie