arduino/ArduinoCore-samd #337: Bootloader modified for SAMD21J17A (128k, 16k) not found by BOSSA

andrewjfox, 2018-07-28:
Hi, I have a custom board based on the zero, but with smaller SAMD21J17A (128k/16k) device instead of the 18A used in the zero.

I've modified the bootloader for the different memory config but can't get the BOSSA flasher to recognize the board. Is there something I'm missing.

New board is "dialler"
- added definition to board_definitions.h, and board_definitions_dialler.h file 
`#elif defined(BOARD_ID_dialler)
  #include "board_definitions_dialler.h"
#else
`
- board_definitions_dialler.h has all the relevant pin settings for the custom board
- changed makefile so it has correct target
`
BOARD_ID?=dialler
NAME?=samd21_sam_ba

CFLAGS_EXTRA=-D__SAMD21J17A__ -DBOARD_ID_$(BOARD_ID)
`

- set the Atmel Studio device to SAMD21J17A

- copied the original linker file to new file bootloader_samd21x17.ld, and make sure makefile points to new linker file
LD_SCRIPT=bootloader_samd21x17.ld

- in linker script changed memory defs to match the 17A device
  FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 0x2000 		
  RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 0x00004000-0x0004 

- in sam_ba_monitor.h set the interface to USB only
`
define SAM_BA_INTERFACE    SAM_BA_USBCDC_ONLY
`

- the USB device enumerates OK, and using an ICE debugger can confirm that it's entering the bootloader mode. It reaches sam_ba_monitor_run() and stays in while(1) loop. 
`
if (main_b_cdc_enable)
    {
      sam_ba_monitor_init(SAM_BA_INTERFACE_USBCDC);
      /* SAM-BA on USB loop */
      while( 1 )
      {
        sam_ba_monitor_run();
      }
    }

`

I can also confirm that the device appears as Arduino Zero bootloader COM port in Windows 7, however whenever I run BOSSA/bossac it doesn't find the device. If I specify the com port it still returns unable to connect to device. I've tried several versions of BOSSA from 1.5 to 1.8 but all have same error.

What am I missing?? If the BOSSA can find the zero samd21x18 device, surely it must be able to detect the samd21x17 device, so I'm assuming there's an issue in the bootloader but can't find it. Is there a setting somewhere in the code that's memory size dependent? Or is there somewhere that makes an assumption that it'a using the x18 variant rather than generic samd21xxx device. I've looked through the code but can't see anything obvious.

I'm beyond frustrated and ready to hurl out the window. Both the device and my stomach contents. If someone can help I'd be truly grateful.

mattairtech, 2018-07-30:
Bossac must also be modified. I added support for the samd21x17a, see https://github.com/mattairtech/BOSSA. You can download the binary (Windows) from https://www.mattairtech.com/software/arduino/bossac-1.7.0-mattairtech-3-mingw32.tar.gz. I also added support in the bootloader, so if you have problems, you might want to also see https://github.com/mattairtech/ArduinoCore-samd. I sell boards with the 17a's, so this is tested.

andrewjfox, 2018-07-30:
Great, will test it out first thing in the morning. Also am curious as to whether the command line and gui versions are in sync or is it best to think of them as seperate packages? Will the gui version also have samd21x17a support soon?

cmaglie, 2018-08-08:
The desktop IDE uses the same bossac as command line.
Nonetheless if you need an updated version of bossac we must distributte a copy of it through the board manager of the IDE and this can happen only after an official release of bossac (or you can provide your own build by forging a custon 3rd party package_index.json).

I'm closing this one as it seems to be bossac related.

closed 2018-08-08 by cmaglie