arduino/ArduinoCore-samd #339: RawHID support

PTS93, 2018-07-31:
It would be great to get the HID device capabilities extended by raw HID. Raw HID is a great way for very robust bidirectional communication with a USB host application.
This is implemented for Teensy and by NicoHood for older AVR Arduinos. It would be incredibly helpful to have this functionality implemented for the SAMD line as well.

For some extended reasonsing as to why this is an important feature. See my post in the CircuitPython HID repo: https://github.com/adafruit/Adafruit_CircuitPython_HID/issues/25#issuecomment-409014365

https://github.com/PaulStoffregen/cores/tree/master/usb_rawhid
https://github.com/NicoHood/HID/tree/master/src/SingleReport

cmaglie, 2018-08-08:
Hi @PTS93 

we are going to add PluggableUSB support on SAMD, there are chances to reuse the library from @NicoHood.

PTS93, 2018-08-08:
That would be great! Can I track the development of that feature somewhere?

adammunich, 2018-09-27:
You can use the original pluggableUSB compatible HID library to define any report descriptor you like. I have an example if you would like to see it.

PTS93, 2018-09-27:
Sure, would definitely be interesting!

NicoHood, 2018-09-27:
This could work now: https://github.com/NicoHood/HID

adammunich, 2018-09-27:
Here ya go https://github.com/adammunich/customHID

PTS93, 2018-09-28:
@NicoHood thank you, that is really awesome!

@adammunich thanks for sharing!