arduino/ArduinoCore-samd #345: Lot of open/uncommented PRs...

tuxedo0801, 2018-08-07:
Hey there,

just wanted to point out to @cmaglie / @sandeepmistry / @facchinm / whom it may concern that there are a lot of PRs, and many of them are open since months or even years. And also many of them are just commented by other users. There's no comment from the arduino-core-samd team. 

This is somehow frustrating. We're trying to make the ArduinoCore somehow better, but we're somehow not heared. 

How do you want to proceed with this? 

br,
Alex

cmaglie, 2018-08-20:
Hi @tuxedo0801 

first of all thank you so much for your interest and **love** :-) we really appreciate that (of course not only from you but from all our awesome contributors).

Your concerns are real, we can only confirm that some PR / issue gets stale, waiting for review.

I can only see that this is endemic in many big/successful open source software and in Arduino in particular. In Arduino case this is mainly due to lack of developers-time and to the incredible amount of inputs we receive every day. This is so massive that we have to prioritize it, so patches or improvements that have small impact on our user base are going to be placed very low in our backlog.

Just to give you a vague idea, I personally receive 200+ notifications each day only from github, and those are not spam or something that I can skip easily, I have to actually **read** and digest all of them. Just reading will take at least one hour of my daily job if I'm lucky (but it's not rare that I need 2x or 3x that if I start to answer only to **some** of them...).

Beware, I'm not trying to justify myself or my team, I just want you (and the other contributors) to know that we are not deliberately ignoring issues but the actual reason behind the lack of responsiveness is just because we are constantly under a DDOS attack :-).

rocketscream, 2018-09-23:
@cmaglie @mbanzi I would have to agree with rest here. Some of the issues here has been sitting for ages even thought solution and ideas were proposed by users but mostly went unanswered. Was about to file in a new issue, but when I see the amount of them that went unanswered, it just droves me away.

PaulStoffregen, 2018-09-23:
I agree this is an ongoing problem, but solutions are not so simple.  I experience this same problem on a much smaller scale with Teensy.  There's never enough time to properly review every request & potential contribution.

This may sound harsh, but the unpleasant reality is many contributions and suggestions have serious problems, often not immediately obvious.  Some seemingly "trivial" things can bring large unintended consequences.  Most people make a contribution that's they've tested on only a single board.  Often it's only been tested in the scope of their particular application.

With the IDE, there are nightly builds (actually updated hourly), a special beta with farther-future features, and on each pull request arduinobot provides ready-to-run builds of the whole IDE with that particular change.  Despite all these ways anyone in the Arduino community can easily download and run a test version of the IDE, consider the 1.8.7 release was needed only 3 weeks after 1.8.6 to fix several issues that had gone unnoticed for months of those beta & nightly tests.

It's easy to see unanswered issues and wish for a better process.  I'm sure there are ways to improve, and I hope progress will be made.  But I believe it's important to keep in mind the reality of Arduino, with about a million active users running the IDE on multiple platforms, targeting dozens of different hardware products and hundreds of 3rd party boards, thousands of libraries, across an incredibly broad range of end use applications!

lxrobotics, 2019-09-16:
While I can agree with the sentiments voiced by @tuxedo0801 we are making an honest effort to drastically reduce the number of open issues (Merge Week Initiative :smirk: ) which is why I am closing this PR. Regardless of this I want to stress that community input via issues is always welcome - creating a meaningful PR is even better.

closed 2019-09-16 by lxrobotics