arduino/ArduinoCore-samd #347: bootloader makefile issue .elf filename missing

better-adapted, 2018-08-16:
related file - ArduinoCore-samd\bootloaders\zero\makefile
platform : windows 10 / Atmel studio / arduino zero

issue : when debugging the elf output file was just getting called ".elf" and debugging would not begin

so I had to add the new line below "BOARD_ID?=arduino_zero" @ line 60
NAME?=samd21_sam_ba

[Makefile.txt](https://github.com/arduino/ArduinoCore-samd/files/2294225/Makefile.txt)

facchinm, 2018-08-20:
@better-adapted we never noticed it because we normally run https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/build_all_bootloaders.sh#L5 , whould you mind opening a PR with that extra line? Thanks!

facchinm, 2018-08-24:
Solved by merging #351, thanks!

closed 2018-08-24 by facchinm