arduino/ArduinoCore-samd #358: No usbcore support for isochronous or interrupts?

adammunich, 2018-09-22:
It appears there is no usbcore support for isochronous endpoints or firing interrupts in pluggableUSB classes? This makes audio really tough. 

Should we start by adding 

`
	else if (config == USB_ENDPOINT_TYPE_ISOCHRONOUS | USB_ENDPOINT_IN(0))
	{
		usbd.epBank1SetSize(ep, 64);
		usbd.epBank1SetAddress(ep, &udd_ep_in_cache_buffer[ep]);
		usbd.epBank1SetType(ep, 0x2); // ISOCHRONOUS IN
	}
	
	else if (config == USB_ENDPOINT_TYPE_ISOCHRONOUS | USB_ENDPOINT_OUT(0))
	{
		usbd.epBank1SetSize(ep, 64);
		usbd.epBank1SetAddress(ep, &udd_ep_out_cache_buffer[ep]);
		usbd.epBank1SetType(ep, 0x2); // ISOCHRONOUS OUT
	}
`

To initEP()?

adammunich, 2018-09-23:
I've begun digging into this and noticed some things which ought to be changed in SAMD21_USBDevice.h

In example, DoubleBufferedEPOutHandler assumes that the buffer size is always 64, and, that the endpoint should be bulk always. Wouldn't moving usbd.epBank0SetType(ep, 3) to InitEp in usbcore.cpp make a bit more sense?

As well, changing usbd.epBank0SetSize(ep, 64); to usbd.epBank0SetSize(ep, size);

```
DoubleBufferedEPOutHandler(USBDevice_SAMD21G18x &usbDev, uint32_t endPoint, uint32_t bufferSize) :
		usbd(usbDev),
		ep(endPoint), size(bufferSize),
		current(0), incoming(0),
		first0(0), last0(0), ready0(false),
		first1(0), last1(0), ready1(false),
		notify(false)
	{
		data0 = reinterpret_cast<uint8_t *>(malloc(size));
		data1 = reinterpret_cast<uint8_t *>(malloc(size));

		usbd.epBank0SetSize(ep, 64);
		
		// Adam Edit
		// usbd.epBank0SetType(ep, 3); // BULK OUT
		// End Adam Edit

		usbd.epBank0SetAddress(ep, const_cast<uint8_t *>(data0));

		release();
	}
```

adammunich, 2018-09-24:
Good to hear @facchinm.

I have begun working on this in a way that should require minimal modifications to the USBcore -- I am trying to find the best intersection of efficiency and compatibility. 

I have thus far been able to create an isochornous endpoint + a feedback endpoint, and break out the start_of_frame interrupt through pluggableUSB in a way that doesn't break other pluggableUSB libraries. the next step is to calculate a real value for the feedback endpoint.

I will keep this thread updated as I make further progress.

adammunich, 2018-09-25:
Another strange thing: USBSetup packet only captures the first stage of the transaction. 

`// Endpoint 0 Received Setup interrupt
	if (usbd.epBank0IsSetupReceived(0))
	{
		usbd.epBank0AckSetupReceived(0);

		USBSetup *setup = reinterpret_cast<USBSetup *>(udd_ep_out_cache_buffer[0]);

		/* Clear the Bank 0 ready flag on Control OUT */
		// The RAM Buffer is empty: we can receive data
		usbd.epBank0ResetReady(0);
`

Is where setup is cached from the output buffer, but you'll notice that the packet ends at wLength. It contains the header for the setup packet, but none of the data specified by wLength!

`// Low level API
typedef struct {
	union {
		uint8_t bmRequestType;
		struct {
			uint8_t direction : 5;
			uint8_t type : 2;
			uint8_t transferDirection : 1;
		};
	};
	uint8_t bRequest;
	uint8_t wValueL;
	uint8_t wValueH;
	uint16_t wIndex;
	uint16_t wLength;
} USBSetup;
`

In consideration that control packets are not supposed to be more than 64 bytes I am going to add the remaining bytes as an array to this packet so that control commands such as Audio Mute can be interpreted.

adverity, 2018-11-27:
I've added Ischonronous IN, to create a 'usb audio mic' like applciation.  For 48k, 16bit audio you need 192 byte per ms, This means you have to make the udd_ep_in_cache_buffer larger... which currently means making all the cache_buffers larger - perhaps this could be done on a malloc per endpoint ?  

```
#define EPX_SIZE 256 // 64 for Full Speed, EPT size max is 1024

static __attribute__((__aligned__(4))) //__attribute__((__section__(".bss_hram0")))
uint8_t udd_ep_out_cache_buffer[7][EPX_SIZE];

static __attribute__((__aligned__(4))) //__attribute__((__section__(".bss_hram0")))
uint8_t udd_ep_in_cache_buffer[7][EPX_SIZE];
```
 
(I don't think you can do 3x64bytes transferes because you only get 1 isonchronous token from the host per 1ms (I think this correct)).

I've added another send function with callback on data transfer compelte,

`typedef void (*epTransferCompleteCallback)(uint8_t ep);`

```

static epTransferCompleteCallback epTransferCallback[7] =
{
	NULL,
	NULL,
	NULL,
	NULL,
	NULL,
	NULL,
	NULL
};

uint32_t USBDeviceClass::send(uint32_t ep, const void *data, uint32_t len, epTransferCompleteCallback callback)
{
	epTransferCallback[ep] = callback;		
	usbd.epBank1EnableTransferComplete(ep);
	send(ep, data, len);
}
```
and added the callback from the handleEndpoint...
```
void USBDeviceClass::handleEndpoint(uint8_t ep)
{
#if defined(CDC_ENABLED)
	if (ep == CDC_ENDPOINT_IN)
	{
		// NAK on endpoint IN, the bank is not yet filled in.
		usbd.epBank1ResetReady(CDC_ENDPOINT_IN);
		usbd.epBank1AckTransferComplete(CDC_ENDPOINT_IN);
	}
	if (ep == CDC_ENDPOINT_ACM)
	{
		// NAK on endpoint IN, the bank is not yet filled in.
		usbd.epBank1ResetReady(CDC_ENDPOINT_ACM);
		usbd.epBank1AckTransferComplete(CDC_ENDPOINT_ACM);
	}
#endif

#if defined(PLUGGABLE_USB_ENABLED)
	if(epTransferCallback[ep])
	{
		epTransferCompleteCallback callback = epTransferCallback[ep];
		epTransferCallback[ep] = NULL;
		callback(ep);				
	}	
#endif
}
```
in the code, I just schedule the first transfer from the setup, and then the next from the callback..
```
void audiocallback(uint8_t ep)
{
	USBDevice.send(0x4, audiobuf, 192, audiocallback); 
}
```

mweetman3, 2020-03-17:
Was there any further progress made on this? I am trying to implement the suggested changes.
Thanks for information, It will be great to get this working.

abaelen, 2020-04-26:
I just managed to write up a library that enables isochronous IN transactions **without** **touching the ArduinoCore**. It attaches the isochronous interface using PluggableUSB and relays the UDD_Handler to my own code in order to manage the configuration of the Isochronous endpoint (outside USBCore as no support for >64byte). I also have written the code for receiving the isochronous data transfer on Windows machine using Console C++ application based on the WinUSB library. Wireshark shows a transfer of 10,000 x (1023+51)byte packages in 10secs. It is fairly raw code (based on the USBCore library), but could be good starting point growing it more mature. I will share the code if interested? In such case let me know how to best share it. Do I add it here or through my gitHub or via Arduino project or ...?

mweetman3, 2020-04-26:
That sounds REALLY great. I got my application working, but need a few lines added to USBCore.cpp for isochronous endpoint support. The MKZero now shows up as a USB Audio Microphone. I can live with the 64K limitation for my application. Sounds like you did a better job! I think the community it crying out for proppper USB audio support.

abaelen, 2020-04-27:
agree, for those strolling along and curious for  the code, have left it on my github in the meanwhile.