arduino/ArduinoCore-samd #364: "CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later"

ChippyH, 2018-10-10:
I'm seeing the error message "CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later" when trying to use the Board Manager to install 1.6.19  of  Arduino SAMD Boards (32-bits ARM Cortex-M0+ by Arduino

I've uninstalled and reinstalled the IDE and even pasted the .zip file into the /Arduino15/staging/packages folder directly.    Nothing seems to be working :-(

Any thoughts?

ChippyH, 2018-10-11:
Hi All!  It is working now.  I took my laptop home and deleted / reinstalled the staging folder.
I think what may be happening (although I'm not sure what's happening behind-the-scenes) is that my work network was blocking the Board Manager from communicating with the online JSON file which I'm guessing contains the CRC value.   On my home network with less restrictions it worked fine.

Noted also - once I got the boards installed, my computer was having issues connected to the Circuit Playground Express so I had to uninstall all unnecessary drivers.. then everything worked out great  :-)  Thanks!

facchinm, 2018-10-11:
Glad it worked! If you are behind a proxy at work you can set it up under File/Preferences -> Network tab .

closed 2018-10-11 by facchinm

smtechchauhan, 2019-10-02:
CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
we are facing this problem last 48 hours.
please suggest me the immediate solution for the same.
i shall be thankful to you.

we facing this problem when we are downloading node esp8266 board manager.
plz help

[image]
[image]

facchinm, 2019-10-02:
@mattiabertorello can you provide the instructions to debug the issue via log4j saved logs ?

mattiabertorello, 2019-10-02:
@smtechchauhan you will find the logs in the folder `/Arduino15/logs` could you send us?

jasonkiptoo, 2022-10-24:
CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.what could be the issue
 when installing ESP 8266