arduino/ArduinoCore-samd #365: Using SDU with different pins

gjt211, 2018-10-13:
I would like to be able to use SDU with the Adafruit Feather M0 WiFi board and am not able to get it working as the WINC1500 WiFi module already uses pin 4 for it's chip select.

After lots of searching I believe I need to build execute build.sh in the SDUBoot folder but when I run it (on my Mac) I get bundles of errors and I don't know what to do next.

Can someone provide a quick guide how to build this please?

```
Error: Must specify exactly one sketch file
cat: /Users/gjt211/Downloads/ArduinoCore-samd-master 2/libraries/SDU/extras/SDUBoot/build/SDUBoot.ino.bin: No such file or directory
```

gjt211, 2018-10-27:
I am not sure the correct place to add this info, but it may help others that want to do what I have done.

Here is part of my code that lets me download the firmware and write it to the SD card. This code won't work as-is, there are other functions that make the complete program.

The boolean variable `fwu_available` is set to true when I receive a firmware filename from a subscribed MQTT topic. the variable `fwu` is a char array that contains the actual firmware filename. This allows me to tell it what firmware version to download. The download is performed over HTTP so it is not secure. This is only for testing.

I also check the size of the downloaded file to ensure it is larger than 80000 bytes then it gets renamed to UPDATE.BIN and then a reset it issued.
```
// This must be called frequently from my main() function.
void check_for_firmware_update(void){
  if (fwu_available == true){
    fwu_available = false;
    Serial.print("\nFirmware Update\n");
    Serial.print("Connecting too update server: ");
    if (WebClient.connect(WebServer, 80)) {
      Serial.println("connected");
      // Make a HTTP request:
      WebClient.print("GET /firmware/ms330/");
      WebClient.print(fwu);
      WebClient.println(" HTTP/1.1");
      WebClient.println("Host: mywebserver.com");
      WebClient.println("User-Agent: MS330-http-Update");
      WebClient.println("Connection: close");
      WebClient.println();
      fwu_download_start = true;
      theFile = SD.open(fwu, O_WRITE | O_CREAT | O_TRUNC);
      if (!theFile) {
        Serial.println("Could not create file");
      }
      
    } else {
      Serial.println("connect failed");
      fwu_download_start = false;
    }
  }

  if (fwu_download_start == true){
    if (WebClient.available()) {
      char c = WebClient.read();
      if (c == '\n' && currentLineIsBlank){
        // end of HTTP header, now save requested file
        while (WebClient.connected()) {
          // stay in this loop until the file has been received
          if (WebClient.available()) {
            c = WebClient.read();  // get file byte
            theFile.write(c);   // save file byte
            //Serial.print(c);
          }
        }
      }
      // detect the end of the incoming HTTP header
      if (c == '\n') {
        // starting a new line
        currentLineIsBlank = true;
      }
      else if (c != '\r') {
        // got a character on the current line
        currentLineIsBlank = false;
      }
        
    }
    // if the server's disconnected, stop the client:
    if (!WebClient.connected()) {
      Serial.println();
      Serial.println("Disconnecting.");
      WebClient.stop();
      theFile.close();
      ls();
      Serial.println("Finished writing to file");
      fwu_download_start = false;

      Serial.print("File size = ");
      Serial.println(theFile.size());

    // My firmware is always larger than 80000 bytes so check the size here
      if (theFile.size() < 80000){
        Serial.print(F("Error: Invalid File Size, removing..."));
        if (!SD.remove(fwu)) {
          Serial.println("failed");
        } else {
          Serial.println("success");
        }
      } else {
        if (!SD.rename(fwu, "UPDATE.BIN")) {
          Serial.println("File rename error");
        } else {
          Serial.println("File renamed OK");        
          // following should go in the renamed ok section
          delay(1000);
          Serial.println(F("==> Rebooting"));
          NVIC_SystemReset();
          delay(1000);
        }
      }
    }
  }
}
```

SolveEtCoagula07, 2024-10-30:
@gjt211 I am having similar issues as you had.  Any chance you'd be willing to provide some assistance?