arduino/ArduinoCore-samd #366: Problems related to USB device support

AlanStern, 2018-10-15:
I just saw the two commits from January and August related to the DoubleBufferedEPOutHandler class.  In fact this area still has several remaining issues:

When a SET_CONFIGURATION request is received, the code deletes the bulk-OUT endpoint's existing handler (along with its buffers) and creates a new one.  This is wasteful; it would be better to continue using the old handler and simply reinitialize the endpoint.

Also, deleting the existing buffers means that positive values previously returned by available() would become invalid.  Continuing to use the same handler and buffers avoids this problem.

The code seems to assume that the host will send a SET_CONFIGURATION request only when the device is enumerated following each attach.  This is not always true; the host can send SET_CONFIGURATION at any time and some hosts do this as a weak form of reset.  When this happens the host expects the device to set the endpoints' data toggles back to 0, but the existing code does not do this.  (The host also expects the device to clear the endpoints' HALT feature; this is a little less of an issue because nothing in the existing code ever sets the HALT feature, but it still should be addressed.)

In the same way, hosts sometimes use SET_INTERFACE as a very weak form of reset.  The existing code doesn't implement this request correctly at all, which perhaps isn't surprising since it doesn't implement alternate settings for interfaces.  Still, the device shouldn't pretend to handle a request and then ignore it.  Fortunately the USB spec allows a device to STALL a SET_INTERFACE request if alternate settings aren't supported; that's what the code should do here.

The code that handles SET_FEATURE, CLEAR_FEATURE, and GET_STATUS requests for endpoint HALT is completely bogus.  It doesn't even try to look at the endpoint number in the SETUP packet!

Attached is a proposed patch that addresses all these issues except the last.  There are a few other odd things still remaining, such as why DoubleBufferedEPOutHandler::handleEndpoint() should need to use "synchronized" and the reason for including USBDeviceClass::stall().  We can discuss them later.
[usb-fixes.txt](https://github.com/arduino/ArduinoCore-samd/files/2479863/usb-fixes.txt)

PaulStoffregen, 2018-10-15:
> the host can send SET_CONFIGURATION at any time and some hosts do this as a weak form of reset

Any chance you could be more specific about which hosts do this and how to reproduce the condition to make it happen?  I've not seen this before.  I'd really like run some tests on various boards, if I can manage to reproduce this multiple SET_CONFIGURATION behavior.

AlanStern, 2018-10-15:
On Mon, 15 Oct 2018, Paul Stoffregen wrote:

> Any chance you could be more specific about which hosts do this and how to reproduce the condition to make it happen?  I've not seen this before.  I'd really like run some tests on various boards, [...]

I don't know any specific situations where SET_CONFIGURATION is used by
an existing host driver as a type of reset.  However, Linux allows
users to issue SET_CONFIGURATION requests by hand, simply by writing
the bConfigurationValue number to the appropriate device file, and I
have used this on occasion to reset various devices (not a SAMD21,
though).

Here's an example showing how to do it.  I just plugged a SAMD21 device
(an Adafruit Feather M0 board) into my Linux PC, and the following
information appeared in the kernel log:

[11869.345885] usb 1-1.4: new full-speed USB device number 4 using ehci-pci
[11869.431560] usb 1-1.4: New USB device found, idVendor=239a, idProduct=800b, bcdDevice= 1.00
[11869.431571] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11869.431578] usb 1-1.4: Product: Feather M0
[11869.431584] usb 1-1.4: Manufacturer: Adafruit
[11869.480754] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[11869.481661] usbcore: registered new interface driver cdc_acm
[11869.481663] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

The USB device path (which indicates the bus number and port numbers of 
intervening hubs) is 1-1.4.  So by doing (as root):

	echo 1 >/sys/bus/usb/devices/1-1.4/bConfigurationValue

I can tell the device to install configuration 1 (which is the only 
config available on this device -- although you could specify 0 to tell 
the device to unconfigure itself and go to the Addressed state).

Alan Stern