arduino/ArduinoCore-samd #381: Arduino bootloader for SAMD21J18

DasExperimentator, 2019-01-04:
Trying to compile Arduino bootloader
https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero
for use with a SAMD21J18.
The bootloader "as is" works fine with SAMD21G18. But when flashed to J18, the board only works in Programming mode, but is not detected in Native USB mode once a sketch is uploaded. In fact, after I load the sketch, the board simply disappears from Device Manager (not even an Unknown Device)...

I saw a few comments on the Net stating building the bootloader for J18 is "straight forward" and needs almost no modifications. Now, I must be doing something wrong... Can someone share his experience or maybe point me to a relevant guide?

sabas1080, 2019-01-04:
Option 
https://github.com/Microsoft/uf2-samdx1

facchinm, 2019-01-07:
@DasExperimentator I don't understand if the Programming mode you are referring to is "upload via USB" or via serial port. This fork of SAMD core (https://github.com/mattairtech/ArduinoCore-samd) supports 21J directly so you may consider using it.

sandeepmistry, 2019-09-16:
Closing this for now due to lack of feedback. Please re-open with requested info if you are still interested.

closed 2019-09-16 by sandeepmistry

Kprellz, 2021-02-26:
> Trying to compile Arduino bootloader
> https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero
> for use with a SAMD21J18.
> The bootloader "as is" works fine with SAMD21G18. But when [...]

sorry to rehash a closed thread however I’m currently experiencing the exact same issue. Wondering if OP ever found a solution.

DasExperimentator, 2021-02-26:
As far as I recall, we never found a solution to this. Although I admit: we didn't dig much further than what was described in the first post...

Kprellz, 2021-02-26:
Perhaps I can provide some more detail. 

I’ve got a custom board with a samd21j18a trying to burn a bootloader to it using atmel studios and an atmel Ice.
Here’s what I’m doing: 
Open atmel studio, device programmer and read my chip, erase chip, go to fuses and set bootprot to 0, program and verify, load bootloader. I’m using the metro M0 express bootloader although I have tried several other with the same results. program and verify, set boot prot back to 8kbytes. At this point my computer reads the board no problem. Com port shows up, I can get board info from arduino, The whole bit. Then I load any sketch from arduino, even a blank one. It uploads fine and says verify successful. However as soon as it uploads the com port disappears and the device is gone from device manager. No amount of rebooting will fix it the chip is basically bricked at this point. Only way to get it show back up is to repeat the steps of loading the bootloader.

jasani1001, 2021-07-13:
> Perhaps I can provide some more detail.
>
> I’ve got a custom board with a samd21j18a trying to burn a bootloader to it using atmel studios and an atmel Ice.
> Here’s what I’m doing:
> Open atmel studio, [...]

try this
https://github.com/AtmelUniversityFrance/atmel-samd21-xpro-boardmanagermodule/wiki/HowTo:-Install-this-module

I've tested and it works fine for me.