arduino/ArduinoCore-samd #385: digitalWrite ignored after calling analogWrite on PWM pin

don, 2019-01-30:
Calls to digitalWrite don't do anything after calling analogWrite on a PWM pin. Tested on the MKR1000 and MKR WiFi 1010 with SAMD 1.6.20 core.

```
const int ledPin = 5;
unsigned long wait = 10 * 1000;
bool analog = false;

void setup() {
  Serial.begin(9600);

  // Comment the next line to NOT wait for a serial connection for debugging
  while (!Serial);

  // initialize digital pin 5 as an output.
  pinMode(ledPin, OUTPUT);

  digitalWrite(ledPin, HIGH);
  Serial.println("Blinking with digitalWrite");
}

void loop() {
  Serial.println("on");
  digitalWrite(ledPin, HIGH);
  delay(1000);
  Serial.println("off");
  digitalWrite(ledPin, LOW);
  delay(1000);

  if (millis() > wait && !analog) {
    Serial.println("analogWrite will break sketch");
    analogWrite(ledPin, 0xF);
    analog = true;
  }
}
```
The Arduino 101 had a similar issue that was fixed https://github.com/arduino/ArduinoCore-arc32/issues/280

Rocketct, 2019-02-18:
HI @don i have tested the sketch and the issue seems that could be solved simply change the status of the pin by calling the API pinMode(), if you do this after the analogWriting the pin start to work again 

```c++
const int ledPin = 5;
unsigned long wait = 10 * 1000;
bool analog = false;

void setup() {
  Serial.begin(9600);

  // Comment the next line to NOT wait for a serial connection for debugging
  while (!Serial);

  // initialize digital pin 5 as an output.
  pinMode(ledPin, OUTPUT);

  digitalWrite(ledPin, HIGH);
  Serial.println("Blinking with digitalWrite");
}

void loop() {
  Serial.println("on");
  digitalWrite(ledPin, HIGH);
  delay(1000);
  Serial.println("off");
  digitalWrite(ledPin, LOW);
  delay(1000);

  if (millis() > wait && !analog) {
    Serial.println("analogWrite will break sketch");
    analogWrite(ledPin, 0xF);
    analog = true;
    pinMode(ledPin, OUTPUT);
  }
}
```

Rocketct, 2019-03-18:
Closed due to lack of feedback. @don if need reopen the issue

sandeepmistry, 2019-03-18:
Re-opening it as a "known issue" for now.

habazut, 2022-08-10:
It should work the same as on good old Uno. On Uno, digitalWrite() reconfigures a PWM pin back to simple digital mode when called and does undo any special config analogWrite did to the pin output. See as well https://github.com/arduino/ArduinoCore-samd/issues/589

This is a regression when upgrading to a SAMD based board from the Uno.
Regards,
Harald.