arduino/ArduinoCore-samd #394: ATSAMD21E18A support

furieuxjk, 2019-03-24:
Hello!

Is ATSAMD21E18A chip supported with this core?

If yes- what should I choose as a Board if I want to program standalone SAMD21E18A chip?

sabas1080, 2019-03-25:
SAMD21E18A is support, example of board https://github.com/ElectronicCats/Arduino_Boards_Index/tree/gh-pages/electroniccats/samd/variants/bast_pro_mini_m0

furieuxjk, 2019-03-26:
Ok, thanks, I will try this out.

alexwhittemore, 2019-04-01:
To elaborate a little - a few Adafruit boards (Trinket M0, maybe others?) use the E18, and the G18 is extremely similar. Their core fork with their M0 variants is at https://github.com/adafruit/ArduinoCore-samd.

ldab, 2019-04-05:
Bear in mind that Trinket M0 is Crystalless

sandeepmistry, 2019-09-16:
Closing this as answered.

closed 2019-09-16 by sandeepmistry