arduino/ArduinoCore-samd #396: SAM D21 on MKR Zero doesn't work with DS3231 RTC

Bushedo, 2019-03-28:
Have been trying to integrate DS3231 RTC to work with MKR Zero through the I2C bus without success.  Other units on the bus like a BME280 work, and when scanned, the board returns the address of the RTC and its on board memory chip.  Used several DS3231 libraries including RTClib but none seem to work.  After changing the date and time in the code using the example code from the library, the unit always returns the same date 2 Feb 2019 (yes, USA groundhog day!).  The clock module works great with an Uno.  Suspect it could be a timing issue, but can't see any obvious suspects in the library source code.

agdl, 2019-03-29:
Are you sure the module you are using can be supplied at 3.3V?

Bushedo, 2019-03-29:
The RTC is HiLetgo  DS3231 AT24C32.  Seller says it operates from 3.3-5v.
When it is operating with the Uno I’m supplying 3.3V to the module.  The Uno is running on the 5v from the USB cable.  The grounds are tied together.

agdl, 2019-04-03:
If you supply it at 3.3V and using an uno you are using anyways GPIOS on the uno at 5V. So you are stressing the component a lot to use clamping diodes. on the uno you should suply it at 5V

Bushedo, 2019-04-03:
The only reason I was using the Uno was to test the module.  Thought the test would be more realistic if I used the same voltage as the MKR Zero.