arduino/ArduinoCore-samd #401: analogRead on others pin

agdl, 2019-04-13:
Because of this line https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/wiring_analog.c#L133 is not possible to use the ADC channels on pins 2, 3, 11 and on all the MKR boards. Is it really necessary?

sandeepmistry, 2019-04-15:
I think we should leave it as is, `analogRead(A0)` should be the same as `analogRead(0)`.

The variant should be changed to have analog pin numbers to be higher than the digital ones as per the convention in all the cores.

sandeepmistry, 2019-09-16:
Closing this for now due to lack of feedback.

closed 2019-09-16 by sandeepmistry

tedtoal, 2023-07-22:
"The variant should be changed to have analog pin numbers to be higher than the digital ones as per the convention in all the cores."

But the SAMD processors have pins that can be configured as either analog OR digital.