arduino/ArduinoCore-samd #406: Arduino zero bootloader doesn't build in Atmel Studio 7

cjheath, 2019-05-03:
In the last week I installed a fresh download of Atmel Studio 7 (build 7.0.122) and Arduino 1.8.9 with SAM hardware package version 1.6.12, and cloned this repository to build a modified bootloader for our custom SAMD21E17A board (don't ask).

To start, I opened the Zero bootloader solution in Atmel Studio. It fails on the first file because it cannot find include file <sam.h>. Looking at the Makefile, the required -I<include-dir> option is missing - sam.h is not present in any of the paths provided.

Because both Atmel Studio are Arduino have a <sam.h> I would need to experiment... and would probably just run into the next error, and the next.

Would someone who knows what they're doing please fix, test and document the correct way to build these bootloaders?

sslupsky, 2019-05-03:
Try the uf2 bootloader from Microsoft.  Mounts as a drive on your desktop and is compatible with BOSSA.

cjheath, 2019-05-03:
If the project includes an Atmel Studio project file, the project should build using that project file. Is that too much to ask?

The UF2 bootloader is not a solution. This is a commercial product and the client has a modified bootloader to use for field upgrades. I did not make this choice, but I need to update it to the newer code.

Bossa is a bad joke, it has zero exception reporting - any protocol error dumps out with "no device". I've added a bunch of stuff to improve it, PR will be forthcoming. But for field upgrades, the client has their own software.

facchinm, 2019-05-14:
Hi @cjheath ,
I tried importing the bootloader project on AtmelStudio7 and, well, it compiles just fine...
Here's the output 
```
------ Build started: Project: samd21_sam_ba, Configuration: Debug ARM ------
Build started.
Project "samd21_sam_ba.cproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\IEUser\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21\bootloaders\zero\samd21_sam_ba.cproj" (target "Build" depends on it):
	Using "RunCompilerTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Users\IEUser\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21\bootloaders\zero" -f "Makefile" DEBUG=1 all 
		make: Entering directory 'C:/Users/IEUser/AppData/Local/Arduino15/packages/arduino/hardware/samd/1.6.21/bootloaders/zero'
		----------------------------------------------------------
		Compiling bootloader using
		BASE PATH = C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino
		GCC  PATH = C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-
		----------------------------------------------------------
		Creating build folder
		mkdir build
		----------------------------------------------------------
		Compiling board_driver_i2c.c to build/board_driver_i2c.o
		"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m0plus -Wall -c -std=gnu99 -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -g3 -O1 -DDEBUG=1 -D__SAMD21G18A__ -DBOARD_ID_arduino_zero -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_i2c.c -o build/board_driver_i2c.o
		++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
		----------------------------------------------------------
		Compiling board_driver_led.c to build/board_driver_led.o
		"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m0plus -Wall -c -std=gnu99 -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -g3 -O1 -DDEBUG=1 -D__SAMD21G18A__ -DBOARD_ID_arduino_zero -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_led.c -o build/board_driver_led.o
		++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
		----------------------------------------------------------
		Compiling board_driver_pmic.c to build/board_driver_pmic.o
		"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m0plus -Wall -c -std=gnu99 -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -g3 -O1 -DDEBUG=1 -D__SAMD21G18A__ -DBOARD_ID_arduino_zero -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_pmic.c -o build/board_driver_pmic.o
		++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
		----------------------------------------------------------
		Compiling board_driver_serial.c to build/board_driver_serial.o
		"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m0plus -Wall -c -std=gnu99 -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -g3 -O1 -DDEBUG=1 -D__SAMD21G18A__ -DBOARD_ID_arduino_zero -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_serial.c -o build/board_driver_serial.o
		++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
		----------------------------------------------------------
		Compiling board_driver_usb.c to build/board_driver_usb.o
		"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m0plus -Wall -c -std=gnu99 -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -g3 -O1 -DDEBUG=1 -D__SAMD21G18A__ -DBOARD_ID_arduino_zero -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\IEUser\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_usb.c -o build/board_driver_usb.o
		++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
		----------------------------------------------------------
		Compiling board_init.c to build/board_init.o
[... cut ...]
Target "Build" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Avr.common.targets" from project "C:\Users\IEUser\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21\bootloaders\zero\samd21_sam_ba.cproj" (entry point):
Done building target "Build" in project "samd21_sam_ba.cproj".
Done building project "samd21_sam_ba.cproj".

Build succeeded.
========== Build: 1 succeeded or up-to-date, 0 failed, 0 skipped ==========
```
I did literally nothing except double clicking the atsln file in `bootloaders/zero` subfolder.

cjheath, 2019-05-14:
That is bizarre. I'll have a close look at it again tomorrow, thanks for the log, I can cross-check every detail. Can you add the detailed build/version numbers for your Windows, Atmel Studio, and Arduino please?

facchinm, 2019-05-14:
Sure, Arduino SAMD core is at 1.6.21; I'm attaching the Atmel studio installation

```
Atmel Studio 7 (Version: 7.0.1931 - )
© 2015 Atmel Corp.
All rights reserved.

OS Version: Microsoft Windows NT 6.2.9200.0
Platform: Win32NT

Installed Packages: Shell VSIX manifest - 7.0
Shell VSIX manifest
Version: 7.0
Package GUID: e874ffe4-fbe3-4624-9a17-61014ede02d0
Company: Atmel Corporation

Installed Packages: Atmel Start - 1.0.158.0
Atmel Start
Version: 1.0.158.0
Package GUID: F8853255-9C7B-4DC2-8E0F-64D9324AEB0E
Company: Atmel

Installed Packages: Atmel Software Framework - 3.40.0.1184
ASF
Version: 3.40.0
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.40.0
Release Description: ASF - 3.40.0 Release

ASF
Version: 3.39.0
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.39.0
Release Description: ASF - 3.39.0 Release

ASF
Version: 3.38.0
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.38.0
Release Description: ASF - 3.38.0 Release

ASF
Version: 3.37
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.37
Release Description: ASF - 3.37 Release

ASF
Version: 3.36.2
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.36.2
Release Description: ASF - 3.36.2 Release

ASF
Version: 3.36.1
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.36.1
Release Description: ASF - 3.36.1 Release

ASF
Version: 3.36.0
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.36.0
Release Description: ASF - 3.36.0 Release

ASF
Version: 3.35.1
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.35.1
Release Description: ASF - 3.35.1 Release

ASF
Version: 3.34.2
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.34.2
Release Description: ASF - 3.34.2 Release

ASF
Version: 3.34.1
Package GUID: 4CE20911-D794-4550-8B94-6C66A93228B8
Company: Atmel
HelpUrl: http://asf.atmel.com/3.34.1
Release Description: ASF - 3.34.1 Release

Installed Packages: LiveWatch - 2.0.65
LiveWatch
Version: 2.0.65
Package GUID: 7DF6DCFD-2BCA-41C7-9C0E-1B7F606B008E
Company: Atmel

Installed Packages: GdbConsole - 7.0.188
GdbConsole
Version: 7.0.188
Package GUID: 49258291-0FED-4501-881B-6BAA91BEBCA8
Company: Atmel

Installed Packages: Atmel Kits - 7.0.122
Atmel Kits
Version: 7.0.122
Package GUID: 6F4B8FE4-C464-4916-8B43-AC92431C1CDF
Company: Atmel

Installed Packages: AtmelToolchainProvider - 7.0.895
AtmelToolchainProvider
Version: 7.0.895
Package GUID: AtmelToolchainProvider.Atmel.10EF9C74-D8DA-4872-85F5-D8BB3101E245
Company: Atmel

Installed Packages: Data Visualizer Extension - 2.18.723
Data Visualizer Extension
Version: 2.18.723
Package GUID: 25dc067d-df31-4e22-be7f-cc6a77ccc7f3
Company: Atmel

Installed Packages: Microchip Gallery - 8.1.0
Microchip Gallery
Version: 8.1.0
Package GUID: AtmelStudio7ExtensionManager
Company: Microchip

Installed Packages: Visual Assist for Atmel Studio - 10.9.2093.2
Visual Assist for Atmel Studio
Version: 10.9.2093.2
Package GUID: 7997A33C-B154-4b75-B2AC658CD58C9510
Company: Whole Tomato Software
```

About "sam.h", it's also contained in `CMISIS-Atmel` package that we and Atmel (Microchip) ship in different forms; the Arduino one can be found in `C:\Users\yourusername\AppData\Arduino15\packages\arduino\tools` after installing SAMD core via board manager.

cjheath, 2019-05-14:
Many thanks. Now I can look to see where my build environment diverges.

cjheath, 2019-05-15:
I have the exact same Arduino and GCC versions. The client was using SAMD core 1.6.19, and I had that version installed both on OSX and under Windows 10. Only the OSX version had a _sam.h_ under the arduino/tools directory. The Windows version had this one (not present on the OSX install):
C:\Users\cjh\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system\CMSIS\Device\ATMEL\sam.h

I refreshed the 1.6.19 SAMD core on Windows 10 and it created the right file:
C:\Users\cjh\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0\CMSIS\Device\ATMEL\sam.h

I have no explanation of why this file was absent before, it was a vanilla install of the same version on a clean machine.

After updating the Windows 10 version to 1.6.21 I still have both versions of _sam.h_. The bootloader now builds cleanly.

So there are two unknowns here:
1) Why did the original SAMD core install on Windows not have the tools include directory?
2) Why does the OSX SAMD core not have the other (non-Arduino) version?

cjheath, 2019-05-15:
To be precise, the Arduino15 directory on OSX lacks the subdirectory packages/arduino/hardware/samd/1.6.21/system
under which the other sam.h lives. Is this correct and expected?

cjheath, 2019-05-17:
Problem solved, though we don't know the reason. Note that this bootloader, when issued the default erase command by bossac from Arduino 1.8.9, will erase the whole chip including the bootloader, which results in "No device found" from bossac - see this PR for context: https://github.com/shumatech/BOSSA/pull/98

closed 2019-05-17 by cjheath