arduino/ArduinoCore-samd #417: arduino zero bootloader can corrupt fuse and corrupt bootloader

mattvenn, 2019-06-06:
**bug**:

using bossa to erase from offset 0 results in fuses being erased and bootloader broken

**steps to reproduce:**

flash the bootloader:

`openocd -f /home/matt/Arduino/hardware/dygma/samd/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c "telnet_port disabled; init; halt; at91samd bootloader 0; program {/home/matt/Arduino/hardware/dygma/samd/bootloaders/zero/samd21_sam_ba.bin} verify reset; shutdown"`

set fuses to protect the bootloader (bootprot = 2, protect first 8k):
`
openocd -f /home/matt/Arduino/hardware/dygma/samd/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c "telnet_port disabled; init; halt; at91samd nvmuserrow; at91samd nvmuserrow 0xFFFFFC5DD8E0C78A; shutdown"`

run bossa GUI and select 'erase all' option:
[image]

bootloader will now start but disconnect from USB after a short time.
fuses are all set to 1:

0xFFFFFFFFFFFFFFFF

board can be recovered with flashing and resetting fuses, although reserved bits are always 1:

0xFFFFFE5DD9FEC78A

If I comment out either the erase or write functionality in sam_ba_monitor.c the problem doesn't happen. If either are in, the problem happens.
I have tried debugging what is happening, but for some reason I am unable to reliably run gdb. I almost always get this error:

```
Continuing.
at91samd21g18.cpu -- clearing lockup after double fault

Program received signal SIGINT, Interrupt.
0xfffffffe in ?? ()
```

**expected behaviour**

there should not be a way to corrupt fuses or bootloader via the SAM-BA interface the bootloader provides.

cmaglie, 2019-07-10:
Is this happening with the latest version of bossa? It happens also runnin `bossac` (the command line version)? The command line version has the `-d` flag that should do a more verbose output that may help debug the problem.

mattvenn, 2019-10-01:
Sorry for the long reply time. Yes it can also happen with bossac. I've seen the Arduino fork of bossa hard-codes the 0x2000 offset into the utility so it's impossible end in this situation.

elfmimi, 2020-01-25:
I'd say that the both side of the bootloader and bossa is equally responsible for this issue. @cmaglie 
https://github.com/shumatech/BOSSA/issues/99#issuecomment-578387029