arduino/ArduinoCore-samd #418: [Question] Calibration area access is not present on samd51

GMagician, 2019-06-07:
Samd51 has a special reserved are with calibration values, its address is 0x00800080.

> The NVM Software Calibration Area contains calibration data that are determined and written during production test. These calibration values should be read by the application software and written back to the corresponding register.

In this area there are some ADC calibration values that "software" has to load (ADCx BIASCOMP, ADCx BIASREFBUF, ADCx BIASR2R, ...).
Since there is no reference to such memory address in this core then I think that these values are never loaded by anyone...

cmaglie, 2019-06-07:
I think it's done already here:

https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/startup.c#L287

GMagician, 2019-06-07:
mmm samd51 has 2 adc and such code is not present in core I have, coming from platformio (adafruit).
I wrongly though such core was originated from here

GMagician, 2019-06-07:
lol this core si for samd21...now it's all clear to me..sorry

closed 2019-06-07 by GMagician