arduino/ArduinoCore-samd #419: Should provide DMA library

matthijskooijman, 2019-06-09:
The SAMD core supports DMA, but this cannot be easily used from a library or sketch currently. In particular, multiple users of the DMA module need some coordination about what channels are in use, where the descriptor memory is located (I think), when to reset the DMA controller, etc.

Currently, I think there is the [Adafruit_ZeroDMA](https://github.com/adafruit/Adafruit_ZeroDMA) library that does this, but the [I2S library](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/I2S/src/utility/DMA.h) also has a DMA coordinator embedded (so these two will not play together nicely).

A single official library contained in the core (or perhaps an API contained in the core itself) would be good here. This could be a fully featured DMA library, but it might also be useful to have a minimal DMA coordinator library (that e.g. only handles resetting/initializing the controller, allocating channels and descriptors and maybe more, leaving all the channel-specific configuration that can coexist to third party or other libraries).

fontanon, 2019-08-05:
I believe this is related https://github.com/adafruit/ArduinoCore-samd/issues/159

ladyada, 2019-08-05:
we use our zerodma handler library a lot for displays, audio, i2s, etc. it works pretty well so far, and supports m0 and m4.

fontanon, 2019-09-11:
Just to let you know about a functional workaround here: https://github.com/adafruit/Adafruit_ZeroDMA/issues/12

matthijskooijman, 2019-09-11:
That workaround is useful to use I2S, but essentially just replaces the Arduino I2S library with the adafruit one which uses Adafruit_ZeroDMA. This does not generally fix this problem by providing a single official DMA library that can be used by all other libraries.

I suspect that the Adafruit_ZeroDMA library might be a good starting point, since it already has some usage. If not using that one verbatim, an official Arduino DMA library should probably try to be compatible with the Adafruit one as much as possible.

ladyada, 2019-09-11:
yes making an official DMA library like Wire and SPI would be really helpful. @PaulStoffregen was also interested in making something