arduino/ArduinoCore-samd #420: SDU compatibility for ATSAMD21G18A-MU (Seeed Wio Tracker)

marcotoldo, 2019-06-13:
Hello,

I'm trying to use the SDU functionality on a Seeed Wio Tracker board (which has an on-board sd card reader working with the Arduino SD Library) but it seems to be unable to work ...

The board features a ATSAMD21G18A so as far as I can see the SDU library should work since its bootloader should be the same as (or very similar to) Arduino Zero.

Actually the SDU library was not available on Arduino IDE for that board so I simply added it externally, then changed SDU.cpp to be sure the zero booloader is used:

```
unsigned char sduBoot[0x4000] = {
#include "boot/zero.h"
};
```

and included it as in the provided example.

I strictly followed the official example but after resetting the board the sketch is not updated, no matter what (even tried UPDATE.bin in lower case update.bin)

Am I missing something?   Does the SD card needs a particular formatting or the bin file needs to be placed in a specific folder other than root?

Thanks in advance

facchinm, 2019-06-17:
Hi @marcotoldo ,
to make SDU compatible with a custom board you need to recompile `SDUBoot.ino` (in `extra` folder) changing `SDCARD_SS_PIN` to match your board.
`build.sh` script in the same folder performs this action and creates the needed header file, that you can than include in your sketch.
Let me know if it works!

marcotoldo, 2019-06-17:
Hi @facchinm,

thank you very much for your feedback.

I'll surely give a try to your suggestion (we would like to implement an OTA feature for that board and the easiest way we found is trying to use the SDU library after downloading the updated .bin via gprs).

I'll report to you about the trials in the next days.

Best regards

gjt211, 2019-06-17:
Hi @marcotoldo,
This does work, I done this about a year ago for a custom board I made for exactly the same reasons as you.
Regards

marcotoldo, 2019-06-17:
Hi @gjt211,

glad to hear about that!

Best regards

marcotoldo, 2019-06-18:
Hi @facchinm,

thanks for pointing me to the right direction (I'm still a beginner in the Arduino world ;-)

Unfortunately after lots of trials I'm still unable to make it work: as far as I can see the header file with the compiled SDUBoot.ino is not included in the sketch (sketch size doesn't change when SDU.h is included)

It does work for supported boards such as Arduino Zero (sketch size changes as well) but when selecting my custom board adding SDU.h has no effect.

Just for the sake of cleanness I generated the "right" header file by recompiling SDUBoot.ino for my own board even if it should match arduino zero (no need for changes, slave select pin was already equal to 4), added it to the boot directory and forced the SDU library to load it:

```
__attribute__ ((section(".sketch_boot")))
unsigned char sduBoot[0x4000] = {
/*#if defined(ARDUINO_SAMD_ZERO)
  #include "boot/zero.h"
#elif defined(ARDUINO_SAMD_MKR1000)
  #include "boot/mkr1000.h"
#elif defined(ARDUINO_SAMD_MKRZERO)
  #include "boot/mkrzero.h"
#elif defined(ARDUINO_SAMD_MKRFox1200)
  #include "boot/mkrfox1200.h"
#elif defined(ARDUINO_SAMD_MKRGSM1400)
  #include "boot/mkrgsm1400.h"
#elif defined(ARDUINO_SAMD_MKRWAN1300)
  #include "boot/mkrwan1300.h"
#elif defined(ARDUINO_SAMD_MKRWIFI1010)
  #include "boot/mkrwifi1010.h"
#elif defined(ARDUINO_SAMD_MKRNB1500)
  #include "boot/mkrnb1500.h"
#elif defined(ARDUINO_SAM_ZERO)
  #include "boot/mzero.h"
  #error "Unsupported board!"
#endif*/
#include "boot/wio.h"
};
```
but as I found out the header file is not added to the my compiled sketch (though the compiler complains if that is missing ...)

Any hint about what's wrong with that?  Do I have to compile my sketch for one of the supported boards in order to make it work ? (actually that may be painful because many libraries I need are available for my custom board only)

Sorry if I'm asking something very obvious to you but being a beginner I can't figure out what is missing.

Thank you very much

Best regards

facchinm, 2019-06-19:
Are you changing target when compiling for Wio board? The cause my be that some linker script lines are missing. Here are the lines for Zero board
https://github.com/arduino/ArduinoCore-samd/blob/5c4d667dc7c8be0fd378183a591bd90f1b423b65/variants/arduino_zero/linker_scripts/gcc/flash_with_bootloader.ld#L68-L73
Try checking if the Wio `variant` linker script contains these lines; if it does not, adding them will 99% fix the issue :slightly_smiling_face:

marcotoldo, 2019-06-19:
Great!  Adding those lines to the .ld script did the trick :-)

Now it's working like a charm :+1: 

Thank you very much for your support @facchinm , you really saved my day :-)

facchinm, 2019-06-19:
Super, glad it helped :wink: !
Closing the issue as solved

closed 2019-06-19 by facchinm