arduino/ArduinoCore-samd #422: Error burning Zero bootloader using Atmel-ICE

timdriedger, 2019-06-24:
I see the following error when attempting to burn the Zero bootloader using Atmel-ICE with the 1.8.1 samd core...

```
Open On-Chip Debugger 0.10.0+dev-gf0767a31 (2018-06-11-13:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.27.0082
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 400 kHz
Info : SWD DPIDR 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x0000296c msp: 0x20007fb0
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Error: couldn't open C:Users	imdrAppDataLocalArduino15packagesrduinohardwaresamd.8.1/bootloaders/zero/samd21_sam_ba.bin
embedded:startup.tcl:477: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 536
at file "embedded:startup.tcl", line 477
Error while burning bootloader.
```

Rolling back to 1.6.21, burning the bootloader is successful...

```
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-12:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.27.0082
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD: partitioning the first flash page into 8 subpages
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 8192 bytes from file C:\Users\timdr\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21/bootloaders/zero/samd21_sam_ba.bin in 0.890651s (8.982 KiB/s)
** Programming Finished **
** Verify Started **
verified 6504 bytes in 0.562310s (11.295 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

Windows 10 Pro
Version 1903
Arduino 1.8.9

facchinm, 2019-06-25:
Hi @timdriedger ,
thanks for reporting. It looks like an horrible path related bug; could you update again to 1.8.1 and enable verbose upload to read also the invoked commandline? It should resemble something like
```
..../packages/arduino/tools/openocd/0.10.0-arduino7/bin/openocd -d2 -s /home/martino/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7/share/openocd/scripts/ -f ..../packages/arduino/hardware/samd/1.8.1/variants/nano_33_iot/openocd_scripts/arduino_zero.cfg -c "telnet_port disabled; init; halt; at91samd bootloader 0; program ..../packages/arduino/hardware/samd/1.8.1/bootloaders/nano_33_iot/samd21_sam_ba_arduino_nano_33_iot.bin verify reset; shutdown" 
```

Since the backslash seem stripped in your error it could be a unix/windows clash on openocd
```
Error: couldn't open C:Users	imdrAppDataLocalArduino15packages�rduinohardwaresamd�.8.1/bootloaders/zero/samd21_sam_ba.bin
```

facchinm, 2019-06-28:
Closing as fixed by eb62911

closed 2019-06-28 by facchinm

matthijskooijman, 2019-06-29:
@facchinm, I'm a bit confused by your commit (message). It talks about double/triple {}, but I only see double ones? It also talks about a difference between unix and windows, but then uses the same lines for both platforms. Or is it that this difference used to exist but not anymore in recent OpenOCD versions?

Also, it's a bit confusing that there is a double {} now. AFAICS, the inner {} is the Arduino IDE replacement syntax, while the outer one is passed as-is to OpenOCD and is TCL string quoting. Maybe it would be good to add a comment that indicates this, to help future readers understand these lines?

facchinm, 2019-07-01:
Adding a comment would be good :slightly_smiling_face: 
The original "code" was https://github.com/arduino/ArduinoCore-samd/commit/5704a06f781558efe9f141c6323fcae9e64036dd#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9L190 , and openocd 0.9 properly escaped anything in `{{...stuff...}}`.  0.10.0 just quotes the first level, thus we now need to provide just `{... stuff...}`.
The windows vs unix thing meant that the error is not visible if there is no backslash, even if we don't tcl-quote the path.
I'll add a line to explain this in platform.txt :wink: