arduino/ArduinoCore-samd #423: Keyboard and Mouse not working from 1.8.0 release on

per1234, 2019-06-26:
Windows 10 64 bit / Ubuntu 19.04 / Raspbian 9.4
Keyboard library 1.0.2 / Mouse library 1.0.1
MKR1000, MKR Fox 1200, MKR GSM 1400

When compiled with Arduino SAMD Boards 1.8.0, 1.8.1, or https://github.com/arduino/ArduinoCore-samd/tree/284bc5a3185756cf6d107c002cc8841d772d07d4 installed, the following code:
```c++
#include <Keyboard.h>
void setup() {
  pinMode(2, INPUT_PULLUP);
  pinMode(LED_BUILTIN, OUTPUT);
  Keyboard.begin();
}

void loop() {
  if (digitalRead(2) == LOW) {
    Keyboard.print("hello");
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
  }
}
```
has no effect on Windows and gives this sort of incorrect output on my Linux machines:
```
heloooooooooooohelohelohelooooooooooooheloooooooooooohelheloheloooooooooooohelooooooooooooheloooooooooooohhelooooooooooooheloooooooooooohelooooooooooooheloooooooooooohelheloooooooooooohelohhelohelhhhelhelohhe
```
The following code
```c++
#include <Mouse.h>
void setup() {
  pinMode(2, INPUT_PULLUP);
  pinMode(LED_BUILTIN, OUTPUT);
  Mouse.begin();
}

void loop() {
  if (digitalRead(2) == LOW) {
    Mouse.move(0, 200);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    Mouse.move(0, -200);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
  }
}
```
Has no effect on Windows, but seems to work correctly on Linux.

When compiled with Arduino SAMD Boards 1.6.28 or 1.6.21, both sketches work as expected on Windows and Linux.

facchinm, 2019-06-26:
I think the problems on Windows could be due to CDC Serial becoming pluggable, so the descriptor could be "reversed". About the keyboard bug I'll investigate later today. Same for Windows of course

facchinm, 2019-06-26:
I found the culprit as commit  a8d208af70dc92713e1f83639e9ff8ac96a186a7. Reverting looks safe but I want to investigate a bit more.

facchinm, 2019-06-26:
The problems in windows are 2 (related):
* Arduino drivers only specify the CDC as first interface; forcing `USB input device` as target driver just works. The workaround only applies to Windows10 since previous versions has no CDC generic driver.
* Everything is fine if SerialUSB is the first interface. Unfortunately there's no way to tell the compiler about the instantiation order of pluggable devices. 
A new API could be added to force a pluggable module to be the first (or the last).

pedroslopez, 2020-04-16:
any update on this?

facchinm, 2020-04-16:
@pedroslopez are you still experiencing the issue? The main bug should be fixed on latest samd core, but the driver installation is still flawed (on win7)

pedroslopez, 2020-04-17:
Yeah, it doesn't seem to be working for me. I'm running a test with the following code:

```cpp
#include <Keyboard.h>

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Keyboard.begin();
}

void loop() {
  delay(5000);
  Keyboard.print("hello");
  
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);

}
```

I just took the code from the first post of this issue and removed the button. The keyboard is not writing anything. I'm using Arduino 1.8.12, Windows 10 64 bit, keyboard library 1.0.2 with an MKR1000.

facchinm, 2020-04-17:
Ok, so your problem is likely related with drivers installation; a solution is to open Device Manager and uninstall the MKR1000 drivers; after reset, both serial and HID (keyboard) drivers should be automatically installed (thanks to windows 10). The same procedure unfortunately doesn't work on WIn7 because the serial drivers are not included in the operating system.
Let me know if it works!

Workshopshed, 2020-05-24:
I also have problems with the keyboard not working on the Arduino MKR Zero board.  Looking at the drivers in USBDeview, it does look like the HID device has been assigned the serial USB driver?

[image]

Workshopshed, 2020-05-24:
Swapping back to 1.6.21 fixes the issue and the HID device is hooked up correctly.

[image]

[image]

facchinm, 2020-05-25:
Hi @Workshopshed,
when you compile the sketch with core 1.8.x the order of descriptors get reversed, so on Windows 10 (and only there) you must uninstall the drivers shipped by us for the system to correctly installs both the HID and the CDC builtin drivers.  On Win7 the problem is still alive so I'm not closing the issue (and not amending the installation package either)

Workshopshed, 2020-05-25:
So to clarify, there's no upgrade path from 1.6 to 1.8 that would correctly configure the drivers on Windows 10 and we need to work out which drivers uninstall and use a 3rd party tool to remove those?
Or am I missing something?

facchinm, 2020-05-25:
No 3rd party software should be needed (just Win device manager) but if you write an howto it would surely be useful. @per1234 @SimonePDA do we already have an howto somewhere in the website?

The problem with our windows drivers is that they need the CDC descriptor to be the first and they replace the OS driver every time you install an 1.6.x version of the core. So, the first time you recompile on 1.8.x the cached driver will be loaded instead than the HID one. So yes, no clean upgrade path is possible unfortunately :slightly_frowning_face:

akshaygugnani15, 2020-07-07:
You know what you don't require the keyboard.h or 32u4 based Arduino now, I got a solution to use the keyboard and mouse through any Arduino yes you read it right any!

All you have to do is something like this:-

`Serial.println("pressA")`
on the Arduino

And now head to a python script and do this:-

```
import serial
import pyautogui
Arduino_Serial = serial.Serial('COM5', 9600)

while 1:
    incoming_data = str(Arduino_Serial.readline())
    print(incoming_data)
    if 'pressA' in incoming_data:
        pyautogui.press('a')
    incoming_data = ""
```
BOOM YOU ARE DONE

goofyseeker311, 2025-02-27:
I made an IMU mouse using an older arduino studio, that version worked on win11 about 1 year ago, but that version stopped working in the newer win11 versions. I thinks its about the HID drivers, not about the code or the arduino itself.