arduino/ArduinoCore-samd #424: Arduino Nano 33 IoT not listed in Boards menu after update

jamesbebbington, 2019-06-26:
Not sure if this is an issue with the Arduino IDE (v1.8.9) or just user error, but having updated the SAMD boards to v1.8.1, I'm still not seeing the Arduino Nano 33 IoT listed in the Boards menu:

[image]

[image]

I've restarted the IDE, but no change.

facchinm, 2019-06-27:
Hi @jamesbebbington, it may be a known bug of the IDE (@per1234 could probably find the right reference).
If you searched for "nano33iot" in the searchbar and already had SAMD core installed there is not the expected "Update" button but "Install".
This ends up installing 2 versions of the same core and the IDE gets confused.
Manual update procedure:
- navigate to `~/Library/Arduino15/packages/arduino/hardware/samd/`  and manually remove the older core folder
Graphical procedure:
- uninstall SAMD core 1.8.1, you should get the other core as installed, update it without searching anything in the searchbar.

Hope it helps!

jamesbebbington, 2019-06-27:
Many thanks @facchinm, that seems to have been exactly the problem! All fixed now.

closed 2019-06-27 by jamesbebbington