arduino/ArduinoCore-samd #428: pin initialization

sslupsky, 2019-07-03:
Pin initialization appears to default to "INPUT" per PR #28 to maintain compatibility with AVR Arduinos.  It is not clear to me what the consequence is if this was changed.  Could someone please explain why this is necessary?

I would like to suggest that the pins should be initialized according to their definition in the boards corresponding variant.cpp file as follows:

```
// Setup all pins as configured in variant.c
  for (uint32_t ul = 0 ; ul < PINS_COUNT ; ul++ )
  {
    pinPeripheral( ul, g_APinDescription[ul].ulPinType );
  }
}
```

That would ensure that there is no confusion regarding what the intended "default" state of the pins is when the initialization is complete.

sandeepmistry, 2019-09-16:
Hi @sslupsky,

This was done to maintain compatibility with AVR, where the default modes of pins in INPUT.

sslupsky, 2019-09-16:
I see.  Could a compiler directive be used to selectively set the mode to inputs if AVR otherwise, use the variant?

sandeepmistry, 2019-09-16:
Sorry, it's not clear what you mean. I was talking more about existing sketches that run on AVR could run on SAMD boards as is, if they don't explicitly set the pin mode to input.

sslupsky, 2019-09-16:
Ok.  I mistakenly thought the issue was to ensure existing AVR sketches would run on AVR hardware.

Perhaps the best way then is to use an approach similar to what is done with the new VERY_LOW_POWER directive.  However, instead of not initializing the pins at all, initialize them as described by the variant definition (variant.cpp).

The VERY_LOW_POWER directive itself mostly addresses this issue though since the original motivation was to avoid excessive current.

facchinm, 2019-09-17:
Yup, VERY_LOW_POWER defines targets exactly this problem, not by initializing all the pins to the expected function but avoiding any initialization and waiting for the proper user action.

closed 2019-09-17 by facchinm