arduino/ArduinoCore-samd #438: [Question] usbcore

GMagician, 2019-08-08:
I was having issues on Adafruit grand central M4, Marlin was sending chars and they were lost, I checked code and found this:
```c++
if (usbd.epBank1IsReady(ep)) {
			// previous transfer is still not complete
```

is it correct?
or it should be
```c++
if (!usbd.epBank1IsReady(ep)) {
			// previous transfer is still not complete
```

as I found after some search, my issue was in Tinyusb implementation and I fixed it, but I ask here about this doubt.
I didn't dig into code and also my assumption of how usb endpoints work is completely wrong (don't know much about it) but I just read comment (if previous is still not complete I though ep should be not ready)...

sandeepmistry, 2019-09-16:
Hi @GMagician,

Have you observed this behaviour also on a SAMD21 based board? This core only supports the SAMD21 ...

GMagician, 2019-09-17:
Don't have samd21 boards, just samd51. What I report here is mostly a doubt...

closed 2019-10-13 by GMagician