arduino/ArduinoCore-samd #443: Unable to #include <memory> - compilation errors within (MKRZero)

phunanon, 2019-08-18:
Hello,  
I tried importing <memory> to use smart pointers in my sketch (however advisable).  
I've seen that they are used on Arduino ([forum post](https://forum.arduino.cc/index.php?topic=591561.0) mentioning their use), so I was confident I could just include it. I receive these compilation errors:

```
In file included from /home/pat/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/memory:62:0,
                 from /home/pat/.../projectName.ino:1:
/home/pat/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
/home/pat/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
```

The issue can be recreated on Arduino IDE 1.8.9 and an hourly build from today, just by creating an empty sketch and putting  
`#include <memory>`  
at the top, and choosing MKRZero. The issue doesn't occur when picking Arduino Due.

I also apologise if I've filed this issue in the incorrect place.

matthijskooijman, 2019-08-18:
IIRC this also occurs on AVR cores, when you use something like the StandardCPlusPlus library.

I believe the proper fix would be to replace the `min` macro in `Arduino.h` with a template function (template so it still works for all types). Some caveats:
 - It might also need a `#define min min`, so that `#ifdef min` still works (but that might not really be needed).
 - A template function only works in C++, so for C-code, we would need to either keep the macro, or provide just one version of the macro (which I guess must be float?). Or we should drop support for min/max in C.

Also, I guess the proper place to fix this is in https://github.com/arduino/ArduinoCore-API

facchinm, 2019-08-19:
The code in https://github.com/arduino/ArduinoCore-API/blob/namespace_arduino/api/Common.h#L119-L144 should fix both C and C++; we decided not to apply this kind of fixes to the non-API cores but we are late in the porting process, so this one in particular (that happens quite often) could be a good candidate for backporting.